micropython/micropython #4456: SSD 1306 OLED (128x64) shows random dots

sanu-krishnan, 2019-02-03:
I have a custom SSD 1306 OLED (128x64) module connected to Esp32 through I2C.

I am able to find OLED using i2c.scan() method. (Address is 0x3d). 
[An MPU9250 connected to same i2c  also shows in i2c scan with address 0x68]

commands such as `init_display(), poweroff(), poweron(), invert(), contrast()` are working properly.

But I am unable to write texts, shapes etc.. on the screen. Only random dots are appearing.   

Code:
```
import machine, ssd1306
i2c = machine.I2C(scl=machine.Pin(22), sda=machine.Pin(21))
oled = ssd1306.SSD1306_I2C(128, 64, i2c, 0x3d)
oled.fill(0)
oled.text("Hello", 0, 0)
oled.text("Welcome to the", 0, 30)
oled.text("world of IoT", 0, 40)
oled.text(":)", 0, 50)
oled.show()
```

sanu-krishnan, 2019-02-04:
@dpgeorge Thanks for the reply. I checked those parameters and tried to modify the order and values. But didn't work ! But when I manually pulled the reset line of OLED to low for 200ms, everything worked fine.
Is there anyway to do this reset without using hardware reset line ?

OLED used: VG-2864KSWEG49

robert-hh, 2019-02-04:
Now that you talk about it, I had a similar problem with a SSD1106 display. There is a reset command, but that did not work. At the end I used the reset line. If you do not have a spare GPIO for that, you could use an external Reset circuit, like STM1815, STM1816, LM809, MCP809.

sanu-krishnan, 2019-02-05:
OLED is directly soldered on the board, and I would like to make this working without additional reset line. Let me try to work around reset command.

robert-hh, 2019-02-05:
I looked into the datasheet again. I reccalld it wrong. There is not singel reset command. Insetad, you have to set all the different states along to what is mentioned with the physical reset. And in my tests this did not work reliably. Maybe I have done something wrong.

mcauser, 2019-02-13:
I came across some random dots last night while testing a SSD1306 SPI display on the latest `esp8266-20190125-v1.10.bin` firmware. Sounds like an unrelated issue, but worth sharing.

I noticed that if I initialised the `dc` or `rst` pins as inputs, then passed them into the ssd1306 driver constructor, they weren't being reinitialised with `mode=Pin.OUT`.

Calling `dc.value(1)` or `dc(1)` after constructing the driver, then calling `dc.value()` would always return `0`.

Reinitialising the pin with `dc.init(mode=Pin.OUT)` allowed `dc.value()` to return as `1` and `0` when toggled, and subsequent `display.show()`s to draw to the display correctly.

I thought it may have been caused by using pins with hardware 10k pull-ups (GPIO0, GPIO2), so I moved the `dc` and `rst` pins around until it was working. I couldn't reliably replicate the issue.

```
dc = Pin(4)  # was showing random dots, but not always
display = ssd1306.SSD1306_SPI(128, 64, spi, dc, rst, cs)

dc = Pin(4, Pin.OUT)  # works reliably
display = ssd1306.SSD1306_SPI(128, 64, spi, dc, rst, cs)
```

piskernik, 2020-06-01:
Certain boards (e.g. old Adafruit boards with SPI as default) need also a dedicated reset signal, so the __init__ for SSD1306_I2C should look like:
`def __init__(self, width, height, i2c, addr=0x3D, external_vcc=False, res=None):
        self.i2c = i2c
        self.addr = addr
        self.res = res
        self.temp = bytearray(2)
        self.write_list = [b"\x40", None]  # Co=0, D/C#=1
        if res != None:
            res.init(res.OUT, value=0)
            import utime as time
            self.res(1)
            time.sleep_ms(1)
            self.res(0)
            time.sleep_ms(10)
            self.res(1)
        super().__init__(width, height, external_vcc)`

Usage:
`i2c = machine.I2C(-1, scl = machine.Pin(33), sda = machine.Pin(32))
oled = ssd1306.SSD1306_I2C(128, 64, i2c, addr=0x3d, res = machine.Pin(5))`