micropython/micropython #4457: Unix: GCC9 build failures [-Werror=missing-attributes]

hroncok, 2019-02-03:
Since we updated gcc to 9 in Fedora, micropython fails to build. Both 1.9.4 and 1.10 with:

```
gcc -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild -I../../lib/mp-readline -Wall -Werror -Wpointer-arith -Wuninitialized -std=gnu99 -DUNIX -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_USSL=1 -DMICROPY_SSL_AXTLS=1 -I../../lib/axtls/ssl -I../../lib/axtls/crypto -I../../extmod/axtls-include -DMICROPY_PY_BTREE=1 -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0  -DMICROPY_PY_FFI=1 -Os -fdata-sections -ffunction-sections -DNDEBUG  -U _FORTIFY_SOURCE -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMPZ_DIG_SIZE=16  -c -MD -o build/moduselect.o moduselect.c
../../lib/utils/printf.c:105:5: error: '__GI_vsnprintf' specifies less restrictive attributes than its target 'vsnprintf': 'nonnull', 'nothrow' [-Werror=missing-attributes]
  105 | int __GI_vsnprintf(char *str, size_t size, const char *fmt, va_list ap) __attribute__((weak, alias ("vsnprintf")));
      |     ^~~~~~~~~~~~~~
In file included from ./mpconfigport.h:326,
                 from ../../py/mpconfig.h:62,
                 from ../../lib/utils/printf.c:27:
/usr/include/stdio.h:358:12: note: '__GI_vsnprintf' target declared here
  358 | extern int vsnprintf (char *__restrict __s, size_t __maxlen,
      |            ^~~~~~~~~
```

Full build log from 1.10: [build.log.txt](https://github.com/micropython/micropython/files/2825531/build.log.txt)

hroncok, 2019-02-03:
Builds and runs. Any particular Python code I shall try?

pfalcon, 2019-02-03:
make test

hroncok, 2019-02-03:
that passes as part of the build

hroncok, 2019-02-03:
697 tests performed (19289 individual testcases)
697 tests passed
32 tests skipped: builtin_help builtin_range_binop class_delattr_setattr io_buffered_writer namedtuple_asdict sys_getsizeof cmd_parsetree framebuf1 framebuf16 framebuf2 framebuf4 framebuf8 framebuf_subclass urandom_extra urandom_extra_float ure_groups ure_span ure_sub ure_sub_unmatched vfs_basic vfs_fat_fileio1 vfs_fat_fileio2 vfs_fat_more vfs_fat_oldproto vfs_fat_ramdisk vfs_userfs math_factorial_intbig module_getattr mpy_invalid resource_stream schedule extra_coverage

dpgeorge, 2019-02-05:
Thanks for confirming the "fix".  It probably won't do to just delete this line outright because it was added for a reason (although maybe latest uClibc is fixed?).

It might not be possible to get this compiling cleanly for all cases, but at least to start with a simple thing to do would be exclude this declaration for gcc 9 and above:
```c
#if defined(__GNUC__) && !defined(__clang__) && __GNUC__ < 9
int __GI_vsnprintf(char *str, size_t size, const char *fmt, va_list ap) __attribute__((weak, alias ("vsnprintf")));
#endif 
```
That should fix the case reported here and not break any existing builds (assuming they use gcc <9).  If someone one day uses uClibc with gcc 9 and find it doesn't work then further fixes can be made when more info about it is available.

pfalcon, 2019-02-05:
> (although maybe latest uClibc is fixed?).

Maybe it is, but the idea of MicroPython has always been "runs everywhere", and there're still enough "vendor SDKs" in the wild with gcc 3.x and corresponding uclibc. So yes, keeping compatibility with known older cases and patiently waiting until there're usecases of gcc9+uclibc appear makes sense.

> with a simple thing to do would be exclude this declaration for gcc 9 and above:

And ideally, a comment, explaining why test for gcc9 is there.

closed 2019-02-05 by dpgeorge

hroncok, 2019-02-05:
Thank you so much! You rock.

pfalcon, 2019-03-21:
I now see issues re: that weak alias with gcc 8.3.0 (Zephyr toolchain):

https://ci.linaro.org/view/lite-iot-ci/job/lite-aeolus-micropython/716/PLATFORM=96b_carbon,ZEPHYR_GCC_VARIANT=zephyr,label=docker-xenial-amd64-13/console :

~~~
17:14:34 ../../lib/utils/printf.c:125:5: error: expected declaration specifiers or '...' before '__builtin_object_size'
17:14:34  int snprintf(char *str, size_t size, const char *fmt, ...) {
17:14:34      ^~~~~~~~
17:14:34 ../../lib/utils/printf.c:107:5: error: '__GI_vsnprintf' aliased to undefined symbol 'vsnprintf'
17:14:34  int __GI_vsnprintf(char *str, size_t size, const char *fmt, va_list ap) __attribute__((weak, alias ("vsnprintf")));
17:14:34      ^~~~~~~~~~~~~~
~~~

pfalcon, 2019-03-21:
Well, actually it may be just header conflict what triggers cascade of errors, including that one.

dpgeorge, 2019-03-29:
> I now see issues re: that weak alias with gcc 8.3.0 (Zephyr toolchain):

So, is it still an issue, does the check for this need to include down to gcc 8.3?

pfalcon, 2019-03-29:
> So, is it still an issue, does the check for this need to include down to gcc 8.3?

No, it was a different issue.