micropython/micropython #4458: Various unexpeced errors, if initialized timer is not referenced by pythoncode

JohnMTu, 2019-02-03:
When Timer is initialized, but instance is no longer referenced from python code, it's probably subject for GC.
Unfortunatelly after GC there are unexpected errors, when timer should fire.
Most probably underlaying RTOS or c++ code is still trying to reference memory, which was used by python Timer object, which is now gone.

initial topic: https://forum.micropython.org/viewtopic.php?f=2&t=5898&p=33812#p33812

pfalcon, 2019-02-03:
So, be sure to keep references to them.

JohnMTu, 2019-02-03:
yup, that's only way, whichi found working so far.
Perhaps it should be documented.
Better, if can be fixed. Either object prevented to be GCed, or necessary data copied elsewhere, where it's not subject for GC.

pfalcon, 2019-02-03:
It can be fixed in the following way:

~~~
from machine_for_forgetful import Timer
...
~~~

machine_for_forgetful.py:
~~~
import machine

things_to_remember = []

def Timer(*args, **kwargs):
    t = machine.Timer(*args, **kwargs)
    things_to_remember.append(t)
    return t
~~~

closed 2026-05-01 by dpgeorge