micropython/micropython #4459: esp8266 Waiting for STA_IF connect causing reset

astrilchuk, 2019-02-04:
On NodeMCU esp8266 running 1.10 I get a reset after approximately 5 iterations of the while loop shown below.

```
import machine
import network
station = network.WLAN(network.STA_IF)
while not station.isconnected():
	machine.idle()
```

When I reflashed with 1.10 from 1.9.4 it stopped working - but I suspect that may be because it lost saved wifi credentials.

astrilchuk, 2019-02-04:
I can run the code on the REPL and machine.idle() spits out a number each time it is called.

Regardless if I call sleep(0.5) or machine.idle() have I done anything technically wrong here?  I can consistently cause a reset using this code.

robert-hh, 2019-02-05:
Did you try to re-establish the connection by entering station.connect() with your credentials?
this is the standard code I run on every boot, with WIFI_IP, ROUTER_IP, WIFI_SSID and WIFI_PASSWD properly set beforehand.
```
def do_connect():
    import network
    import time

    wlan = network.WLAN(network.STA_IF) # create station interface
    wlan.active(True)       # activate the interface
    if not wlan.isconnected():      # check if the station is connected to an AP
        wlan.ifconfig(config=(WIFI_IP, "255.255.255.0", ROUTER_IP,  ROUTER_IP))
        wlan.connect(WIFI_SSID, WIFI_PASSWD)
        for _ in range(20):
            if wlan.isconnected():      # check if the station is connected to an AP
                break
            print('.', end='')
            time.sleep(0.5)
        else:
            print("Connect attempt timed out\n")
            return
    print('\nnetwork config:', wlan.ifconfig())

do_connect()
```

dpgeorge, 2019-02-05:
> If I replace `machine.idle()` with `time.sleep(0.5)` it works (it waits indefinitely for a connection).

Thanks for testing.  After doing a git bisect it seems this bug was introduced in 869024dd6e62905b7e1069b547856a769b3b24ba (and is nothing to do with the recent power management changes).  More investigation is needed as to why that commit introduced the problem, but I guess that `ets_loop_iter()` is not being called soon enough to feed the WDT after waiting for an extended period in the "waiti 0" instruction.

> Let me emphasize that isn't a big issue for me - I can certainly work around it. But since I found the reset to be easily reproducible, I thought the project might want to know about it.

Thanks for reporting!  It does help to make it better for everyone.

dpgeorge, 2019-02-28:
Should be fixed by 8ce22662feadeb6b564f1087134346cb287251f0

closed 2019-02-28 by dpgeorge

astrilchuk, 2019-02-28:
Thanks for looking at this and for the fix!