micropython/micropython #4466: Need help with implementation

pulkin, 2019-02-05:
Hi everyone,

I started porting micropython to the a9g module (GPRS+GPS) and need help with these and a lot of other questions. The developer docs are nice but too few. My first two questions are:

1. What exactly do I need to implement for Ctrl-D (restart)? Garbage collection? Close all file pointers? Turning off wireless?
2. What is the proper way to sleep? Is there a macro in micropython which respects Ctrl-C?

Your contributions are appreciated [here](https://github.com/pulkin/micropython).

dpgeorge, 2019-02-06:
Thanks for the interest to port MicroPython.  But such questions are out of scope for this issue tracker.  Please use https://forum.micropython.org for these discussions.

closed 2019-02-06 by dpgeorge

pulkin, 2019-02-19:
Unfortunately the forum requires some weird registration process which does not immediately allow posting.

dhylands, 2019-02-20:
> Unfortunately the forum requires some weird registration process which does not immediately allow posting.

No it's just that the first few posts from a new user require moderator approval, so there is a delay between when the post is made and when it shows up. This delay is typically less than a day and often sooner. I think that once somebody has made 3 approved posts, then moderator approval is no longer required. This helps keep the spam down (only the moderators seen the SPAM).

pulkin, 2019-03-08:
Well, it does not work out: waited for a reply for a couple of weeks.

dhylands, 2019-03-08:
> Well, it does not work out: waited for a reply for a couple of weeks.

I'm one of the moderators on the forum and there are no unapproved posts right now.