micropython/micropython #4475: RFC: Create a common usocket_mphal

Jongy, 2019-02-07:
We currently have 4 different `modusocket` implementations, all supposed to expose the exact same interface.

I believe lots of the code can be shared. There's lot of parameters handling and conversion between `mp_obj_t` to simple data types, and vice versa. Most of the systems do expose a BSD sockets API, or, at least, their API can be easily wrapped to achieve the BSD sockets API.

Having multiple implementations of the pythonic API is bad because things go wrong and little inconsistencies show up between different ports. Generally I think the pythonic API should be implemented only once, so having a file like `utime_mphal.c` is a nicer way.

Also, when the python API impl is done, it's easier to now implement it for more ports.

Would `modusocket_mphal.c` do it?

dpgeorge, 2019-02-08:
The problem with unifying the `usocket` implementations in that the ports usually have different underlying implementations with subtle differences that need to be accounted for.  Currently there are 3 broad classes of `usocket` code:
1. `extmod/modlwip.c` which directly wraps the low-level interface provided by lwIP, so that threads are not needed.
2. Those based on BSD sockets, like in unix, esp32 and cc3200 ports.  Note that cc3200 has some differences here to standard BSD sockets.  It's possible that unix and esp32 `usocket` code could be combined.
3. stm32 provides a separate general framework for plugging in external socket providers, eg cc3000 and wiznet (when the TCP/IP stack lives on the external device).  This is similar to (2) above but goes through a dispatch layer so that multiple different BSD socket-like providers can be glued together.

If some kind of unification was done (which is a good thing to consider) then it would be very important to retain the efficiency of how (1) is currently implemented.

dpgeorge, 2019-02-08:
Sorry, closed in error.

Jongy, 2019-02-09:
I'll give it a shot while keeping in mind about efficiency and code size (which tend to get hurt while generalizing stuff). Start with something small and see if we find it useful.

dpgeorge, 2025-10-04:
This is something that still could be done, probably following how the `machine` module classes have been factored out into `extmod/machine_*.c` code.

But since there's been no activity here, I'll close the issue for now.