micropython/micropython #4478: SPI with pyboard

lcha911, 2019-02-07:
Hello,

When sending data from the pyboard SPI lines, my MOSI pin malfunctions in a certain condition. When the last byte is larger than 127 (\x7f). The pin would remain high even after the data transmission is finished.

I have my pyboard v1.1 flashed with the most recent micropython.

dpgeorge, 2019-02-08:
Thanks for the report.  Are you able to provide a short piece of code that reproduces the error, so it can be reproduced by others?

lcha911, 2019-02-08:
Thank you. Below, the communication ends normally:
from pyb import SPI
spi = SPI(1, SPI.MASTER, baudrate=600000, polarity=0, phase=0)
spi.send(b'\xff\x00\xff\x70')

After changing the last byte, the MOSI line stays high after the communication:
from pyb import SPI
spi = SPI(1, SPI.MASTER, baudrate=600000, polarity=0, phase=0)
spi.send(b'\xff\x00\xff\x80')

chrismas9, 2019-02-08:
Is this causing you a problem? I am not surprised that the MOSI line retains the value of the last bit after the SPI CLK stops. The STM32F405 manual shows MOSI as undefined after the last clock. In slave mode it shows that MISO retains the value of the last bit.

Normal SPI peripherals shouldn't care about the bus state before the first clock or after the last clock, except for the short setup and hold times. What peripheral are you using?

lcha911, 2019-02-08:
I was programming EVAL-ADF5356. It didn't work reliably earlier, so I suspected the bug might be from this undefined MOSI state.

The system seems to work now, and the previous bug might be from something else. Thanks for the explanation.

dpgeorge, 2019-02-21:
Yes, it seems that you can't rely on the MOSI pin being anything in particular after a transfer (and retaining the last bit shifted out seems like a reasonable behaviour).  If more control is needed then the MOSI pin can be explicitly controlled after the SPI transfer is done, and then given back to the SPI before the next transfer, eg:
```python
machine.Pin('X8', machine.Pin.ALT, alt=5) # give pin to SPI peripheral
spi.write(...) # do transfer
machine.Pin('X8', machine.Pin.IN, machine.Pin.PULL_UP) # set MOSI as input pull-up
```

closed 2019-02-21 by dpgeorge