micropython/micropython #4479: extmod/moduhashlib.c includes crypto-algorithms/sha256.c for mbedtls builds

tomlogic, 2019-02-08:
I recently upgraded to 1.10 and am using mbedtls for my sha256 implementation.  I noticed that `extmod/moduhashlib.c` was including another sha256 implementation at the end of the file, and think it should be updated to only do so if `MICROPY_SSL_MBEDTLS` is 0.

```
#if MICROPY_PY_UHASHLIB_SHA256 && !MICROPY_SSL_MBEDTLS
#include "crypto-algorithms/sha256.c"
#endif
```

There probably won't be any references to the functions defined there, so the linker will leave them out, but there's no need to include it.

tomlogic, 2019-02-08:
Another possibility is to include that file earlier in the file, before `uhashlib_sha256_make_new()` for non-mbedtls builds.

dpgeorge, 2019-02-08:
This was just fixed in #4469..!

Jongy, 2019-02-09:
Haha I was quick with that.

tomlogic, 2019-02-09:
So funny.  I searched existing open issues and looked at the version of the file from HEAD, but I guess I hadn't fetched recently enough to see #4469.  Closing this out as already fixed.

closed 2019-02-09 by tomlogic