micropython/micropython #4482: esp8266 release includes multiple sample codes in flash & module import priority

barbudor, 2019-02-08:
Hi
I am in the learning curve of µPy and esp8266 and I found accidently that a few sample codes are embedded in the firmware I am currently using : "esp8266-20190125-v1.10.bin"

As I was playing with the sample code `http_server.py`, it took me some time to understand why none of my changes were taken into account : as `http_server.py` is already in the firmware, all my `import http_server` through REPL were loading the flash version instead of the one I was working on.

I found out that -at least- the following samples code are part of the above firmware:

```
from micropython-lib\umqtt.simple
	example_pub_button
	example_sub_led
from micropython\examples\network
	http_client.py
	http_client_ssl.py
	http_server.py
	http_server_ssl.py
```

May I suggest to remove the sample codes from the firmware has it can be quite disturbing for a newbie (and may save a few bytes too) ?

As a side question, is there a reason why prioritizing import of modules from the firmware over user modules of the same name ? I would consider it much more usefull the other way in order to override and test lib updates.

Thanks

peterhinch, 2019-02-09:
This question should be raised in [the forum](http://forum.micropython.org/index.php), not the MicroPython bug tracker.

There is a good reason why these modules are embedded: to save RAM on the ESP8266. It is possible to override the default search order:
```python
>>> import sys
>>> sys.path.append(sys.path.pop(0))
```
Explanation: `sys.path` shows the current search order, with `''` being modules frozen in flash. The above script moves `''` from first to last position in the search order.

pfalcon, 2019-02-09:
> There is a good reason why these modules are embedded: to save RAM on the ESP8266.

No, there's actually even better reason why they're embedded: to allow newcomers a quick and smooth start with MicroPython ESP8266 port, where they can just flash the firmware and already have a selection of samples to play with, before they proceed to the next "question of century" - how to get their own source on the device, followed by hundreds of other "questions of century" typically coming of newbies, majority answered in documentation, on forum, etc.

barbudor, 2019-02-09:
Hi @peterhinch and @pfalcon 
Thanks for the feedback

Peter, this is not a how-to-question but a proposal-for-change. So I believe it is definitively the right place.

Paul, I understand what you mean but as such a newcomer, I wanted to point that the opposite results has been achieved, thus the proposal-for-change.
I've already had much time to learn how to edit, load and run "hello world" py script long before looking into running an http server.
I just wanted to point out to the frustration that can come from uploading and making changes into a sample code and not seing any of the code changes reaching any results. With no disrespect to the team, I don't see the point in embedding a sample that you cannot see the code, make change and evaluate the impact of your changes. Should you embedded sample code, I believe they should be in the file system part, not in the frozen modules.

This might not be seen as a important as a bug report nor as a pull request for fixing or improving the project's code but I trust this is the right place to share it.

Thanks for reading.
Best regards.

closed 2019-02-09 by barbudor

pfalcon, 2019-02-09:
> I wanted to point that the opposite results has been achieved, thus the proposal-for-change.

I doubt that's the right conclusion to make. Instead, it just shows an ancient wisdom, that no matter what the subject is, there will be different opinions about it.

> This might not be seen as a important as a bug report nor as a pull request for fixing or improving the project's code but I trust this is the right place to share it.

Absolutely. It's always interesting to hear first impressions, they are golden, and can pinpoint some issues. But they aren't always "right". I may even agree there's something to do about. But given that there's always a lot to do, I'd let numbers decide priority, and that's the first complain I hear that prepackaging examples is bad thing. If we just remove them, it's not going to lead to bliss: there will a cascade of angry complaints that stuff people relied upon is gone behind their backs.

adamples, 2020-03-14:
If this indeed is a feature a resonable question to ask would be: where is it documented?

In official documentation? No: https://docs.micropython.org/en/latest/search.html?q=http_server
Maybe in README file in repo? No: https://github.com/search?q=org%3Amicropython+http_server&type=Code
At least some post on the forum or official website? No luck here: https://www.google.com/search?q=%22http_server%22+inurl:micropython.org&sxsrf=ALeKk00YQYv3s7F6gJEwVPWBEGefAF15lw:1584189720959&filter=0

It doesn't make sense (to me) to add example code to official distribution if there is no other way to discover it than by accident. Thanks for the workaround though!