micropython/micropython #4484: module wiznet5k server not disconnecting prior to close

t35tB0t, 2019-02-10:
When wiznet5K TCP sockets are being used as servers, client connections hang due to server closing socket without issuing proper FIN-ACK disconnect.  Adding disconnect in the close() method fixes client handshake but breaks the socket __del__ destructor method (system hangs, waits a long time for timeouts).   The wiznet 5K socket disconnect method closes the socket but this command is never called (code pragma'd out).

dpgeorge, 2019-02-21:
Are you using the wiznet5k in the default mode (where the TCP/IP stack is on the wiznet), or in MAC raw mode (where the TCP/IP stack is on the host MCU)?

t35tB0t, 2019-02-22:
Built firmware with wiznet5K in MAC raw mode for PyBoardv1.1 then added 
the following modifications.  This hacked a fix and now we get a 
disconnect when a socket is closed.  The issue now is that deleting the 
socket object also triggers the close method which now includes a 
disconnect that may hang the system for some time out period.  i.e. a 
CTRL-C break will hang until the disconnect times out.  So probably need 
to flow through a proper separation of intentional TCP disconnects 
versus slam-bam connection dropping.

PS - we're also testing LWIP on STM32 RMII and working through bugs with 
with (missing TCP checksums) but that's for another thread not yet 
created...

--- a/ports/stm32/modnwwiznet5k.c
+++ b/ports/stm32/modnwwiznet5k.c
@@ -140,7 +140,10 @@ STATIC void
wiznet5k_socket_close(mod_network_socket_obj_t *socket) {
      uint8_t sn = (uint8_t)socket->u_param.fileno;
      if (sn < _WIZCHIP_SOCK_NUM_) {
          wiznet5k_obj.socket_used &= ~(1 << sn);
-        WIZCHIP_EXPORT(close)(sn);
+        if (socket->u_param.type == Sn_MR_TCP)
+            WIZCHIP_EXPORT(disconnect)(sn);
+        else
+            WIZCHIP_EXPORT(close)(sn);
      }
  }

+++ b/ports/stm32/mpconfigport.h
+#define MICROPY_PY_LWIP             (1)

+++ b/ports/stm32/mpconfigport.mk
+MICROPY_PY_WIZNET5K ?= 5500

dpgeorge, 2019-02-22:
Thanks for the info.

> we're also testing LWIP on STM32 RMII and working through bugs with with (missing TCP checksums) but that's for another thread not yet created...

You may want to hold off on this because I'm very close to pushing native Ethernet support for RMII (see #3808).

t35tB0t, 2019-02-22:
To clarify:  We merged in 3808 and are looking into checksum bugs.  
Testing on STM32F429 and STM32F767.  Have made several other mods to 
3808 (ported SMT32F767 native ethernet support to STM32F429).

dpgeorge, 2019-02-22:
> We merged in 3808 and are looking into checksum bugs.

Please see new #4541.  I've done a lot of testing on it and it's quite robust (checksums are computed in the hardware).