micropython/micropython #4489: The "identity" of the object for the "items" method returns incorrect values

fizista, 2019-02-11:
For the micropython we have:
```python
MicroPython v1.9.4-782-geb446ec22 on 2019-01-31; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> a = {}
>>> print('ID: ', id(a.items()) == id(a.items()))
ID:  False
>>> print('ID: ', a.items() == a.items())
ID:  False
```
However, for Python 3.5 we have:
Python 3.5.6 (default, Jan 31 2019, 09:36:50) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> a = {}
>>> print('ID: ', id(a.items()) == id(a.items()))
ID:  True
>>> print('ID: ', a.items() == a.items())
ID:  True

Is this a mistake or maybe a property of the micropython?

pfalcon, 2019-02-11:
> Is this a mistake or maybe a property of the micropython?

The latter.

fizista, 2019-02-11:
Thanks. Good to know .

closed 2019-02-11 by fizista