micropython/micropython #4499: webrepl hangs when putting files of size 0

Jongy, 2019-02-14:
I'm using `webrepl_cli.py` to sync my code to the esp32. When uploading `__init__.py` files which are typically  empty, it hangs.

```
$ touch x
$ python webrepl_cli.py x <ip>:/x
...
Sent  0 of 0 bytes
```

A few minutes later, killing the script leads to a print in the UART: `dupterm: EOF received, deactivating`.

I'd say either `webrepl_cli.py` or `modwebrepl.c` don't handle that `0` very well/

dpgeorge, 2019-02-21:
Thanks for the report, I can confirm this bug.  It looks like it's just at the modwebrepl.c end, that it needs extra logic to handle a file of size 0.

dpgeorge, 2019-02-28:
Should be fixed by 9b2a97a903f0841a75947d86d49f0056d43f81b0

closed 2019-02-28 by dpgeorge