micropython/micropython #4500: uhashlib not compatible with micropython-hmac

Jongy, 2019-02-14:
`micropython-hmac` uses `micropython-hashlib`, which attempts to import various hash classes (`sha1`, `sha256` etc) from `uhashlib` if they exist. For any missing class, it imports a pure Python implementation.

Now, the pure Python differ a bit in the API they expose, comparing to the implementations in `uhashlib` itself. That wouldn't be of a problem, but the `hmac` module later uses some of these APIs... So if your `uhashlib` has e.g `sha1` then you can't use `hmac(digestmod=sha1)`.

Specifically, `digest_size`, `block_size` and `copy` are missing (at least from what I've seen, possibly more missing attributes exist)

dpgeorge, 2021-05-25:
IMO `micropython-hmac` should be changed so that it doesn't need to make a copy of the underlying hash object.

mvo5, 2021-07-21:
Fwiw, I ran into the same issue and was wondering if https://github.com/micropython/micropython-lib/pull/435 is the right way to fix this. I.e. leave uhashlib alone but instead when someone imports the python-stdlib hashlib give them the pure-python versions first which have the required {digest,block}_size and copy() helpers.

jonnor, 2024-09-19:
I believe that https://github.com/micropython/micropython-lib/pull/515 has fixed this incompatibility.

closed 2025-02-04 by projectgus