micropython/micropython #4502: Ethernet support on the Olimex ESP32-POE

XenGi, 2019-02-14:
I'm trying to get ethernet working on the ESP32-POE from Olimex. It's a ESP32 with a LAN8720A chip.

According to https://github.com/micropython/micropython-esp32/issues/172 the similar boards ESP32-GATEWAY and ESP32-EVB seem to work. I tried the same code on the ESP32-POE but without luck.

Here are the schematics of the 3 boards:
  - [ESP32-GATEWAY][esp32-gateway]
  - [ESP32-EVB][esp32-evb]
  - [ESP32-POE][esp32-poe]

As you can see the GATEWAY and EVB are quite similar. The POE has a different power layout because of the power over ethernet capabilities but also GPIO00 is not connected to CLKIN anymore. Instead GPIO17 is used which was defined as the Power Pin in the other boards.
I'm not sure what to make out of this. I'm not an electrical engineer so I have no clue what to look for. I tried the [ESP-IDF example demo][demo] and it worked. Micropython seems to do the same thing in the C code below so I'm not sure why it doesn't work.

```
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5060
load:0x40078000,len:8788
ho 0 tail 12 room 4
load:0x40080400,len:6772
entry 0x40081610
I (428) cpu_start: Pro cpu up.
I (428) cpu_start: Application information:
I (428) cpu_start: Compile time:     12:32:34
I (430) cpu_start: Compile date:     Feb 14 2019
I (436) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996
I (442) cpu_start: Single core mode
I (447) heap_init: Initializing. RAM available for dynamic allocation:
I (453) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (460) heap_init: At 3FFB92B0 len 00026D50 (155 KiB): DRAM
I (466) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (472) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (478) heap_init: At 40092834 len 0000D7CC (53 KiB): IRAM
I (485) cpu_start: Pro cpu start user code
I (55) cpu_start: Starting scheduler on PRO CPU.
OSError: [Errno 2] ENOENT
MicroPython v1.10-98-g4daee3170 on 2019-02-14; ESP32 module with ESP32
Type "help()" for more information.
>>> import machine
>>> import network
>>> lan = network.LAN(mdc = machine.Pin(16), mdio = machine.Pin(17), power = None, phy_type = network.PHY_LAN8720, phy_addr=0)
I (22904) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
>>> print('ifconfig :', lan.ifconfig())
ifconfig : ('0.0.0.0', '0.0.0.0', '0.0.0.0', '0.0.0.0')
>>> print('status : ', lan.status())
status :  None
>>> print('isconnected : ', lan.isconnected())
isconnected :  False
>>> print('active : ', lan.active())
active :  False
>>> lan.active(True)
E (55444) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0x0000
E (56444) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0x0000
E (56444) emac: Initialise PHY device Timeout
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: ethernet enable failed
>>>
```

I hope someone with more knowledge about the underlying C code or the harware can shine some light on this.

[esp32-gateway]: https://github.com/OLIMEX/ESP32-GATEWAY/raw/master/HARDWARE/Hardware%20revision%20C/ESP32-GATEWAY_Rev_C_Color.pdf
[esp32-evb]: https://github.com/OLIMEX/ESP32-EVB/raw/master/HARDWARE/REV-D/ESP32-EVB_Rev_D.pdf
[esp32-poe]: https://github.com/OLIMEX/ESP32-POE/raw/master/HARDWARE/ESP32-PoE-hardware-revision-C/ESP32-PoE_Rev_C_Color.pdf
[demo]: https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ESP-IDF/ESP32_PoE_Ethernet

readylan, 2019-02-15:
Hello ! 
I have the same problem. I have contacted with Olimex support and they have send me this line of code:
```
lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), power = machine.Pin(17), phy_type = network.PHY_LAN8720, phy_addr=0)
```
but it doesn't work. They told me to than t is better to post in the MicroPython’s forums since here, the team members are familiar with their boards ..

YonasJ, 2019-02-15:
Hi, I am also having the exact same issue as described above, my board is the Olimex ESP32-GATEWAY rev c, which seem to have the same ethernet wiring as the OP's ESP32POE. My code is: 
`
import machine
import network
lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), power = machine.Pin(17), phy_type = network.PHY_LAN8720, phy_addr=1)

print('ifconfig :', lan.ifconfig())
print('status : ', lan.status())
print('isconnected : ', lan.isconnected())
print('active : ', lan.active())

lan.active(1)
`

And I get the output:
`
rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5056
ho 0 tail 12 room 4
load:0x40078000,len:8788
load:0x40080400,len:6772
entry 0x40081610
[0;32mI (437) cpu_start: Pro cpu up.[0m
[0;32mI (437) cpu_start: Application information:[0m
[0;32mI (437) cpu_start: Compile time:     11:04:40[0m
[0;32mI (439) cpu_start: Compile date:     Feb 15 2019[0m
[0;32mI (445) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996[0m
[0;32mI (451) cpu_start: Single core mode[0m
[0;32mI (456) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (463) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (469) heap_init: At 3FFD7BD0 len 00008430 (33 KiB): DRAM[0m
[0;32mI (475) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM[0m
[0;32mI (481) heap_init: At 40078000 len 00008000 (32 KiB): IRAM[0m
[0;32mI (487) heap_init: At 40092A30 len 0000D5D0 (53 KiB): IRAM[0m
[0;32mI (494) cpu_start: Pro cpu start user code[0m
[0;32mI (64) cpu_start: Starting scheduler on PRO CPU.[0m
main.py: cpu freq: 240.00 Mhz
main.py: mem_free: 121.49
MicroPython v1.10-91-g66f0afc91-dirty on 2019-02-15; ESP32 module with ESP32
Type "help()" for more information.
exec(open('test.py').read(),globals())
E (23865) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
ifconfig : ('0.0.0.0', '0.0.0.0', '0.0.0.0', '0.0.0.0')
status :  None
isconnected :  False
active :  False
E (24865) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0xffff
E (25865) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0xffff
E (25865) emac: Initialise PHY device Timeout
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<string>", line 10, in <module>
OSError: ethernet enable failed
>>> 
`

This, open issue, seems related as well. 

https://github.com/micropython/micropython-esp32/issues/239

And, this seems to be a solution on the C side, but as far as I can tell it should be implemented.

https://github.com/OLIMEX/ESP32-EVB/issues/11

YonasJ, 2019-02-15:
This must be related to something I am doing, when I try it with: esp32-20190113-v1.9.4-779-g5064df207.bin I am able to get it to work. But based on my own build from the trunk, it does not work.

petrkr, 2019-02-17:
Seems same like what I solving now on my board with LAN8720 too. Because I do not want use external clock crystal but use internal PHY's clock, I need switch "clock mode". Unfortunately micropython does not have this part implemented.

I just tried original master with same result as you guys.
```
>>> lan.active(1)
E (23711) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0xffff
E (24711) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0xffff
E (24711) emac: Initialise PHY device Timeout
```

Anyway long story short, if you will hard-code clock mode of PHY to value `3` (regarding to ESP_IDF this stands for `ETH_CLOCK_GPIO17_OUT = 3  /*!< RMII clock output from GPIO17 */` and recompile it.

then it start work (as I expected).

```
>>> lan.active(1)
I (14591) emac: emac reset done
I (14591) network: event 20
True
>>> I (18591) network: event 22
I (19581) event: eth ip: 10.200.0.137, mask: 255.255.255.0, gw: 10.200.0.1
I (19581) network: event 24
```

I can provide current mater BIN for your test (if you want) and will try implement configuration option for this parameter and will try do Pull Request whenever I will get time. Meanwhile hard-coded version for me is enough :)

Petr

petrkr, 2019-02-17:
And I forgot. Here is patch

```
diff --git a/ports/esp32/network_lan.c b/ports/esp32/network_lan.c
index fba4de73a..258c6f3fd 100644
--- a/ports/esp32/network_lan.c
+++ b/ports/esp32/network_lan.c
@@ -146,6 +146,9 @@ STATIC mp_obj_t get_lan(size_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_ar
     config.gpio_config = init_lan_rmii;
     config.tcpip_input = tcpip_adapter_eth_input;
 
+    // Hard coded test for GPIO17 clock output
+    config.clock_mode = 3;
+
     if (esp_eth_init(&config) == ESP_OK) {
         self->active = false;
         self->initialized = true;
```

readylan, 2019-02-17:
Ok, very good, I will try it soon and tell you what happens. Thank you!

petrkr, 2019-02-17:
> Ok, very good, I will try it soon and tell you what happens. Thank you!

you can try that PR already (if you are able compile it yourself). https://github.com/petrkr/micropython/commit/472c5a2937f01fa5543b3caeefc6e0a838080c7f

readylan, 2019-02-17:
> I can provide current mater BIN for your test

Hello, I'm interested in that way ..

petrkr, 2019-02-17:
Can you try https://petrkr.net/micropython-v1.10-108-gbd1b537be-dirty-build-petrkr-LANclock.bin
It's latest `master` with that mine PR, version with optional `power` too. Since you using internal clock, you do not need use power at all, and specify `power = None` is littlebit useless I think, so I made it as default if not specified

Also you have to try if your phy is 0 or 1, for example mine wavewhare module is 1

readylan, 2019-02-18:
Hello, thank you for your help but my board doesn't work:
(phy_addr=0/1)

```
lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), phy_type = network.PHY_LAN8720, phy_addr=1)
lan.active(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: ethernet enable failed
```
I don't know if (or how) can provide more information.

petrkr, 2019-02-18:
You should specify also clock_mode parameter. I guess in your board case it would be:

`lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), phy_type = network.PHY_LAN8720, phy_addr=1, clock_mode=network.ETH_CLOCK_GPIO17_OUT)`

or `phy_mode=0`

readylan, 2019-02-18:
Tested without luck, I don't know what to do ..

petrkr, 2019-02-18:
Regards to schematic (if you have Olimex PoE board) they also have "power" pin at GPIO12. So try

`lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), power=machine.Pin(12), phy_type = network.PHY_LAN8720, phy_addr=0, clock_mode=network.ETH_CLOCK_GPIO17_OUT)`

readylan, 2019-02-18:
I can not get it to work but there are news: the board has link status, its ethernet leds (green and yellow) are active. I've noticed that after unplug from usb and replug again. But I get the same error when trying to activate from the python shell.

petrkr, 2019-02-18:
Maybe you can try skip power= and power-up that module by yourself before lan instance by something like

```
lan_power = machine.Pin(12, machine.Pin.OUT)
lan_power.value(1)
```

and then without `power`
```
lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), phy_type = network.PHY_LAN8720, phy_addr=0, clock_mode=network.ETH_CLOCK_GPIO17_OUT
```

you have to determine if lan_power is value 1 or 0 to power-up that module. As you wrote, you have to see link/activity LED's lighting

also now I check datasheet, that module is 8710, not 8720, can be that also problem? Have you tried some other example? Like from Arduino-C?

readylan, 2019-02-18:
First, I confirm lan_power on 1 power-up the board and leds, otherwise they remain off.
The sketch in Arduino that works for me is [this](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720_internal_clock/ETH_LAN8720_internal_clock.ino).
(changing mdc/mdio pins)

petrkr, 2019-02-18:
One more question, on Micropython, do you have some `boot.py` or `main.py` which can sits on Ethernet GPIO pins ? For me onetime I had problem my old code in boot.py initialized one of PHY pin to OUTPUT and as HIGH or something...

Also when you try different `lan = network.LAN()` you have to reboot board (full, not just CTRL+D) because that old configuration will stay used (like phy_addr, clock mode,etc). If you play with network, it's better always do hard-reboot to be sure all MCU state was reset to default state.

you can use either `machine.reset()` or unplug/replug power

readylan, 2019-02-18:
Ok, you are right, it is working now !!
Thank you very much for your support and sorry for my intermittent status .
```
import network
>>> import machine
>>> lan_power = machine.Pin(12, machine.Pin.OUT)
>>> lan_power.value(1)
>>> lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), phy_type = network.PHY_LAN8720, phy_addr=0, clock_mode=network.ETH_CLOCK_GPIO17_OUT)
>>> lan.active(1)
True
>>> lan.ifconfig()
('192.168.1.170', '255.255.255.0', '192.168.1.1', '192.168.1.1')
```
It is important for me since this board is quite good price and being PoE It is very suitable for my projects.
Thank you again !

petrkr, 2019-02-18:
Good, since you already made it work, you can now also test that if `power=` option will work instead that separated `lan_power`. So you do not need do extra steps...

Regards, also thanks for test

readylan, 2019-02-19:
Yes, it is not neccesary to "prepare power". This works too:
```
>>> import network
>>> import machine
>>> lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), power=machine.Pin(12), phy_type = network.PHY_LAN8720, phy_addr=0, clock_mode=network.ETH_CLOCK_GPIO17_OUT)
>>> lan.active(1)
True
>>> lan.ifconfig()
('192.168.1.170', '255.255.255.0', '192.168.1.1', '192.168.1.1')
```
Regards,

coon42, 2019-02-20:
Works fine with Olimex ESP32-POE :). Thanks for the patch!

XenGi, 2019-02-21:
Nice work guys. Thx to all of you. I'm getting more of the ESP32-POEs now. :)

dpgeorge, 2019-02-21:
The `clock_mode` parameter was added in 7d8c71c222ed7793c99792eb15966504e039dae4

closed 2019-02-21 by dpgeorge

adrian-vlad, 2021-10-03:
Hi, sorry for commenting on an old issue, but it seems that `clock_mode` parameter has been removed from the firmware and I didn't manage to make the LAN work.

I have an ESP32-POE-IND, Rev E board and I flashed the latest firmware, esp32-20210902-v1.17.bin, with no luck.
```
>>> import network, machine
>>> lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), power=machine.Pin(12), phy_type = network.PHY_LAN8720, phy_addr=0, clock_mode=network.ETH_CLOCK_GPIO17_OUT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'ETH_CLOCK_GPIO17_OUT'
>>> lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), power=machine.Pin(12), phy_type = network.PHY_LAN8720, phy_addr=0)
E (29017) emac_esp32: emac_esp32_init(308): reset timeout
E (29017) esp_eth: esp_eth_driver_install(198): init mac failed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: esp_eth_driver_install failed
```

How can I use the wired LAN on this board without rebuilding the firmware myself?

petrkr, 2021-10-05:
I made issue for that on espressif long time ago as they drop this change from runtime and kept only during build time

https://github.com/espressif/esp-idf/issues/6564

But they already closed that issue, so now it's about integrate to micropython build and use this SDK which supports that

hagenbuch, 2022-02-16:
I have tried this with ESP32-POE (rev. B) and micropython-1.18

> True
> ('192.168.1.170', '255.255.255.0', '192.168.1.1', '192.168.1.1')

but I only get the error

Attribute Error: 'module' object has no attribute 'ETH_CLOCK_GPIO17_OUT

Where did I go wrong?

I'm installing with Thonny on Linux Mint 20.3, went fine as far as I can see.
Andreas

Benoit3, 2022-04-19:
Hello,
have a look here
https://www.olimex.com/forum/index.php?topic=8684.0

I've done my own build based on current master.
[esp32-poe-iso.zip](https://github.com/micropython/micropython/files/8516188/esp32-poe-iso.zip)

Just use:
lan = network.LAN(mdc = [machine.Pin](https://machine.pin/)(23), mdio = [machine.Pin](https://machine.pin/)(18), power = [machine.Pin](https://machine.pin/)(12), phy_type = network.PHY_LAN8720, phy_addr = 0)

Benoit

hagenbuch, 2022-06-16:
Fantastic, many thanks, Benoit! However, are these changes included in the 1.19 version that was issued yesterday? Did you happen to offer a pull request?

I know what a compiler is :) but I have to confess I feel a bit overwhelmed right now..

Cheers
Andreas

commit 7d8c71c2 (2019-02-21): esp32/network_lan: Add arg to constructor to set clock mode for ETH PHY.

This optional parameter for network.LAN clock_mode can be used for cases
where the clock source is different from the default GPIO0.  Fixes #4502.