micropython/micropython #4505: 'I2C' object has no attribute 'writeto'

zxc807492312, 2019-02-15:
there is error on ssd1306.py
i2c seems doesn't have those functions
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ssd1306.py", line 99, in __init__
  File "ssd1306.py", line 36, in __init__
  File "ssd1306.py", line 61, in init_display
  File "ssd1306.py", line 104, in write_cmd

closed 2019-02-15 by zxc807492312

MichealRick, 2019-03-02:
My mistake is the same as yours. Have you solved it?Share it with me.

IHOXOHI, 2020-04-24:
me too, ... and I don't known if this mistake is mine...

jimmo, 2020-04-24:
Could you please raise this on the [forum](https://forum.micropython.org/) (and include more information, such as which board you're using and your code)

peterhinch, 2020-04-24:
This is probably because the I2C object was created using `pyb` instead of `machine`.

robert-hh, 2020-04-24:
In case this was a serious question: The driver was updated a few days ago to deal with uncommon display ratios. I used it since then and it works. And yes, as Peter said, you have to use I2C from the machine module.

IHOXOHI, 2020-04-25:
>  'writeto'  (#4505)

Hi,
Thanks for your help, and your diligency.
I will post on micropython forum...with the same login.
See you soon.

I use the original pyboard.
my code:
from machine import I2C
i2c = I2C( 2 )
import ssd1306
lcd = ssd1306.SSD1306_I2C( 128,32, i2c )

IHOXOHI, 2020-04-25:
Hi Peter,

Thanks for your help and your disponibility.

My code is :

from machine import I2C
i2c = I2C( 2 )
####the i2c.scan() command return [60] value
import ssd1306
lcd = ssd1306.SSD1306_I2C( 128,32, i2c )
###at this step I have error about no "writevto" parameter with this object
##I have tried to switch writevto by writeto without success

Thanks for all,
Have a fun,
Sellegri Jerome.

IHOXOHI, 2020-04-25:
Hi Robert,

Thanks for your help.

I use the machine module without success...

The worst is that I can't use too the lcd160 original display of the pyboard...

My previous uses of pybaord were corects with classicals buzzer, led buttons, lights sensor...

Thanks a lots,
Sellegri Jerome

peterhinch, 2020-04-25:
In that case I think your firmware is out of date. On my Pyboard:
```python
MicroPython v1.12-371-gf534b9976-dirty on 2020-04-16; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
>>> from machine import I2C
>>> i2c = I2C(1)
>>> 'writevto' in dir(i2c)
True
>>> 
```

IHOXOHI, 2020-04-26:
Hi Peter,

I enjoy your help.

Now, my firmware is up to date.
But the problem persist.

The result of your commande about the i2c return the same result than your... I have "True".
But when I initialize my lcd, I have yet the same error "Errno 19 ENODEV"...

Thanks...
Have a fun.

peterhinch, 2020-04-27:
This means that the device isn't being detected. Possibly a hardware or wiring problem.