micropython/micropython #4511: ESP8266 1.10 polling causing crash

astrilchuk, 2019-02-15:
```
import usocket as socket
import uselect as select

addr = socket.getaddrinfo('0.0.0.0', 8000)[0][-1]
s = socket.socket()
s.setblocking(False)
s.bind(addr)
s.listen(1)

poll = select.poll()
poll.register(s, select.POLLIN)

res = poll.poll(10000)
```

mcauser, 2019-02-16:
Duplicate of #4508

astrilchuk, 2019-02-16:
Sorry about the duplicates - I'm not sure what happened :-(

dpgeorge, 2019-02-18:
Thanks for the report.  It's quite a bad bug, introduced in 29da9f06709df77d15437b67db5978ecafcccaef

It should now be fixed by 42c0e440b9635fb33c0b2a162f5bfe3122dce079

To workaround the error please use a backlog of 2 or more, eg: `s.listen(2)`.

dpgeorge, 2019-02-18:
> To workaround the error please use a backlog of 2 or more

By that I mean that if you can't upgrade to the latest firmware (which fixes the bug) and need to use v1.10, then use a backlog of 2 or more.

closed 2019-02-18 by dpgeorge