micropython/micropython #4512: couldn't build firmware for nrf port

rlnktt, 2019-02-16:
I am trying to build the firmware for nrf port , following the instructions given at

https://github.com/micropython/micropython/tree/master/ports/nrf#micropython-port-to-the-nordic-semiconductor-nrf-series

with no success after executing the command

"make" inside the "ports/nrf" directory

[image]

glennrub, 2019-02-17:
Hi @rlnktt ,

I'm not seeing this on Linux, so i wonder if this is an issue with the linker in OSX. Could you tell me which linker you are using, and the version number of it? I'm not having a Mac myself, but a bit more input would be very helpful to try to find the issue.

Meanwhile, waiting for a proper solution, you can try to bypass it by hard coding the values for the nrf52832. Edit the _boards/memory.ld_ and set the values to the following:
```
MEMORY
{
    FLASH_TEXT (rx) : ORIGIN = 0x0, LENGTH = 512K     /* app */
    RAM (xrw)       : ORIGIN = 0x20000000, LENGTH = 64K
}
```

rlnktt, 2019-02-18:
@glennrub Thanks for the quick reply there!! Thought of hard-coding. But somehow wanted a proper solution. 

What you suggest seem to be meaningful.

Let me give it a try and update you :)

rlnktt, 2019-02-18:
@glennrub Thanks again!!

This temporary patch actually fixed the issue!!

But how do I find the version of the linker. Please advice

glennrub, 2019-02-20:
> But how do I find the version of the linker. Please advice

@rlnktt , assuming you are using GNU Arm Embedded Toolchain, please issue the following command:
`
arm-none-eabi-ld -v
`
This will tell more about which version of the linker you have, and will help narrow it a bit down.

aykevl, 2019-03-01:
I've seen this error before, most likely `arm-none-eabi-ld` is too old. Releases from the last ~3 years should have support.

This one definitely works:

```
$ arm-none-eabi-ld --version
GNU ld (2.28-5+9+b3) 2.28
Copyright (C) 2017 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

rlnktt, 2019-03-08:
> > But how do I find the version of the linker. Please advice
> 
> @rlnktt , assuming you are using GNU Arm Embedded Toolchain, please issue the following command:
> `arm-none-eabi-ld -v`
> This will tell more about which version of the linker you have, and will help narrow it a bit down.

Lakshmis-MacBook-Pro:nrf lnsri22$ arm-none-eabi-ld -v
GNU ld (GNU Tools for ARM Embedded Processors) 2.23.2.20140731

is what I get

rlnktt, 2019-03-08:
> I've seen this error before, most likely `arm-none-eabi-ld` is too old. Releases from the last ~3 years should have support.
>
> This one definitely works:
>
> ```
> $ arm-none-eabi-ld --version
> GNU ld [...]

Lakshmis-MacBook-Pro:nrf lnsri22$ arm-none-eabi-ld -version
GNU ld (GNU Tools for ARM Embedded Processors) 2.23.2.20140731
Copyright 2012 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

is what I get

aykevl, 2019-03-08:
This feature was implemented [4 years ago](https://sourceware.org/bugzilla/show_bug.cgi?id=4643). You are using a binutils version from [6 years ago](https://sourceware.org/ml/binutils/2012-11/msg00215.html). I would suggest updating the toolchain.

rlnktt, 2019-03-08:
@aykevl Thanks a lot for pointing this out. I will update the tool-chain , build the firmware and let you know

dpgeorge, 2019-06-24:
I assume this was resolved by using a more recent toolchain, so closing.

closed 2019-06-24 by dpgeorge