micropython/micropython #4514: ESP32: usocket.socket has unexpeted type "function"

peterhinch, 2019-02-16:
On Unix build (also Pyboard and ESP8266):
```python
>>> import usocket
>>> type(usocket.socket)
<class 'type'>
>>> s = usocket.socket()
>>> isinstance(s, usocket.socket)
True
>>> 
```
On ESP32:
```python
>>> import usocket
>>> type(usocket.socket)
<class 'function'>
>>> s = usocket.socket()
>>> isinstance(s, usocket.socket)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: issubclass() arg 2 must be a class or a tuple of classes
>>> 
```

dpgeorge, 2019-02-18:
Thanks for the report.  According to the docs (CPython) `socket.socket` is a function, although it's actually implemented as a class.  So it's debatable as to whether one can expect `socket.socket` to be a class and use it for things like `isinstance`.

But anyway, I think it's better to be consistent here within MicroPython where all ports implement it the same way, which would be as a class.  So it should be fixed.

dpgeorge, 2019-02-18:
Fixed by 303f27f6568a8e5ba386a2b31c692548c07a4acf

closed 2019-02-18 by dpgeorge

Jongy, 2019-02-18:
Just sayin', it's one of the things #4475 would have automatically done (and prevented in the future) if it was properly implemented.. :)