micropython/micropython #4518: memory allocation error from time to time after machine.reset()

nedoskiv, 2019-02-18:
Hello,
I'm using ESP32 with MicroPython v1.9.4-684-g51482ba92 without PSRAM

I got lot of stuff some _threads, while writing and changing I often restart the board, using machine.reset()

from time to time i got memory allocation error when my scripts start to load. So question is why, board reboot if some of them are too big it should fail every time. Not from time to time. Is there something that I'm missing?

nevercast, 2019-03-06:
I'm going to suspect memory fragmentation being caused by each of the threads loading their own code. Can you try an experiment, put all your imports in your main script before you start all the threads. Then start the threads. Import everything first if you can.

Here is why;
I suspect that when you start each of your threads, they run a function that loads some python code. When python code is loaded from Flash it must be compiled to bytecode before it can run. This compile process creates a lot of memory allocations on the heap. During this compilation process you will have the string contents of the Python code, and the bytecode, both on the heap at the same time. The python code is eventually freed when the compilation is finished but if you are loading code from threads and the scheduler is switching threads you may sometimes be compiling multiple files at once. Causing you to use all your memory (Since you do not have PSRAM).

You can either freeze your code (Compile it in to the firmware), or you can import all your code in one thread so that it is compiled synchronously. 

This is my assumption of what is happening. I cannot be sure though. If this is the case, perhaps MicroPython should not switch threads/tasks when code is being compiled.

nedoskiv, 2019-03-06:
I got all imports in startup script, after that startup script set some common dicts for all threads, and execute them one by one (each thread is started from own .py file) after that script exits and boot.py start my main code (it is short) from main.py

nevercast, 2019-03-06:
Can you import `micropython` and before you start the threads, call `micropython.mem_info(1)`. This will print a memory table to the print output. Can you also do this from each thread, perhaps like this:
```
print('Thread xyz has started')
micropython.mem_info(1)
```

Then can you paste the logs here? I'm interested to see how the memory is being used on the heap.

nedoskiv, 2019-03-07:
```
[0;32mI (6) cpu_start: Starting scheduler on PRO CPU.[0m
[ CONFIG ] loading default configs
I (1191) wifi: wifi driver task: 3ffcc1a8, prio:23, stack:3584, core=0
I (1191) wifi: wifi firmware version: 7aac1f9
I (1191) wifi: config NVS flash: enabled
I (1191) wifi: config nano formating: disabled
[0;32mI (1201) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (1211) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (1221) wifi: Init dynamic tx buffer num: 32
I (1221) wifi: Init data frame dynamic rx buffer num: 64
I (1221) wifi: Init management frame dynamic rx buffer num: 64
I (1231) wifi: Init static rx buffer size: 1600
I (1231) wifi: Init static rx buffer num: 10
I (1241) wifi: Init dynamic rx buffer num: 0
[0;32mI (1291) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0[0m
I (1301) wifi: mode : softAP (30:ae:a4:14:6f:ed)
[0;32mI (1871) network: event 13[0m
[0;32mI (1871) network: event 14[0m
[0;32mI (1871) network: event 13[0m
[0;32mI (1871) network: event 14[0m
[0;32mI (1871) network: event 13[0m
[ WIFI ] Configured in AP mode
[ START ] zone1 initialized!
[ START ] zone2 initialized!
[ START ] zone3 initialized!
[ START ] zone4 initialized!
[ START ] zone5 initialized!
[ START ] zone6 initialized!
[0;32mI (2391) modsocket: Initializing[0m
stack: 1280 out of 15360
GC: total: 111168, used: 27888, free: 83280
 No. of 1-blocks: 159, 2-blocks: 10, max blk sz: 583, max free sz: 4431
GC memory layout; from 3ffe4db0:
00000: h=Thh.MDDDBShDBBBBhhh===h===Shh==h==============================
00400: ================================================================
00800: ================================================================
00c00: ================================================================
01000: =========================================...h==hhSSh==Bh====....
01400: ..S.hh==...S............hS....h=======h=====h===================
01800: ================================================================
01c00: ================================================================
02000: ================================================================
02400: ================================================================
02800: ================================================================
02c00: ================================================================
03000: ================================================================
03400: ================================================================
03800: ===================================================h=======.....
03c00: ........h===ShShShSS...................h==.....DDD...........ShS
04000: ..............................ShBBBBBBB.....h==........B.hShSB..
04400: ......h=====h=====.hSSh=Sh=Sh===================================
04800: ==h=====...hS.......hShS.hS.............hS......Sh==========h===
04c00: ================================================================
05000: ================hShSh...hShSh=..hShS...SSSh=ShS.....hShShShShSh.
05400: h=.h=h==================hh==h=.h==..................ShSD........
05800: ...hh=======.........S................Shh=======================
05c00: =================================.h..........Sh.........ShShh===
06000: ===========......................Sh.........Sh==................
06400: Sh=.............Sh.........Shh=======Fh=.h.....hhhhhhhh=======hh
06800: h...hhhh=======================......hh=========================
06c00: ===============...................................h=======......
07000: ................................................................
07400: .....Fh====================....................................F
07800: ........................................F............F........F.
07c00: .......................................F............F...........
08000: F............F...........F............F.........................
08400: ..........................h=====================================
08800: ===========================Sh........SSh==.........h==..........
08c00: .........................................................h======
[... 437 lines cut ...]
11800: ================================================================
11c00: ====h======.....................................................
       (36 lines all free)
1b000: ....................................
[ WEB ] Waiting for connection, heap 60144

[ START ] system in NORMAL mode!
[ LOG ] 0
[ LOG ] log write 4
 for more information.
>>> 
>>> 
>>> 
```

output is kind a mes, let me explain sequence:

i got 4 thread started, inserted code before start thread and inside thread, here is the sequence:

RLOG,
MFRC522
WEB
SOUND

that ouput is into log file, but a little messy, hope that help, meanwhile I reduce my code a litte and problem almost gone, it happen very rare, guess need to switch to PSram esp32 and have memory like a boss :D