micropython/micropython #4521: build bootloader fails

johncblacker, 2019-02-18:
Trying to build MP.  Got the correct ESP-IDF after looking in Makefile for the correct hash; did a clone --recursive esp-idf; did checkout  5c88c5996dbde6208e3bec05abc21ff6cd822d26; did git submodule update --init --recursive; did make and got the following:
---------------------------  output from make -----------------------------
jobla@DESKTOP-FS36I9G:~/esp32/micropython/ports/esp32$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       /home/jobla/newidf/esp-idf
Current git hash:   5c88c5996dbde6208e3bec05abc21ff6cd822d26
Supported git hash: 5c88c5996dbde6208e3bec05abc21ff6cd822d26
mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/bootloader/subproject/main/
mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/bootloader_support/src/
mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/log/
mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/micro-ecc/micro-ecc/
mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/soc/esp32/
mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/spi_flash/
CC /home/jobla/newidf/esp-idf/components/soc/esp32/rtc_clk.c
CC /home/jobla/newidf/esp-idf/components/soc/esp32/rtc_time.c
CC /home/jobla/newidf/esp-idf/components/soc/esp32/cpu_util.c
CC /home/jobla/newidf/esp-idf/components/bootloader/subproject/main/bootloader_start.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/bootloader_clock.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/bootloader_common.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/bootloader_flash.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/bootloader_init.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/bootloader_random.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/bootloader_sha.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/bootloader_utility.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/efuse.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/flash_qio_mode.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/secure_boot_signatures.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/secure_boot.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/esp_image_format.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/flash_encrypt.c
CC /home/jobla/newidf/esp-idf/components/bootloader_support/src/flash_partitions.c
AR build/bootloader/libbootloader_support.a
CC /home/jobla/newidf/esp-idf/components/log/log.c
AR build/bootloader/liblog.a
CC /home/jobla/newidf/esp-idf/components/spi_flash/spi_flash_rom_patch.c
AR build/bootloader/libspi_flash.a
CC /home/jobla/newidf/esp-idf/components/micro-ecc/micro-ecc/uECC.c
AR build/bootloader/libmicro-ecc.a
LINK build/bootloader.elf
build/bootloader/libbootloader_support.a(bootloader_init.o):(.literal.bootloader_init+0xe4): undefined reference to `rtc_wdt_disable'
build/bootloader/libbootloader_support.a(bootloader_init.o): In function `bootloader_init':
bootloader_init.c:(.text.bootloader_init+0x40b): undefined reference to `rtc_wdt_disable'
build/bootloader/libbootloader_support.a(bootloader_clock.o):(.literal.bootloader_clock_configure+0x14): undefined reference to `rtc_clk_init'
build/bootloader/libbootloader_support.a(bootloader_clock.o): In function `bootloader_clock_configure':
bootloader_clock.c:(.text.bootloader_clock_configure+0x76): undefined reference to `rtc_clk_init'
collect2: error: ld returned 1 exit status
Makefile:813: recipe for target 'build/bootloader.elf' failed
make: *** [build/bootloader.elf] Error 1
------------------------- end of output paste -------------------------------------------------------------
Please help, don't know how to proceed.  I'm trying to build so that i have sx1276 module support.  If it's already in a pre-built binary, please point me to it.
Thanks.
John

dpgeorge, 2019-02-19:
It's very strange that the hash is the same but it says it doesn't match... what OS is the build running on?

Please do a "make clean" and then "make V=1" which will give more info about the build.

johncblacker, 2019-02-19:
building on ubuntu(wsl) under win 10.

johncblacker, 2019-02-19:
Here is the build attempt you suggested:

=========================================== pasted output below ===================================================

jobla@DESKTOP-FS36I9G:~/esp32/micropython/ports/esp32$ make clean

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.

** WARNING **

The git hash of ESP IDF does not match the supported version

The build may complete and the firmware may work but it is not guaranteed

ESP IDF path:       /home/jobla/newidf/esp-idf

Current git hash:   5c88c5996dbde6208e3bec05abc21ff6cd822d26

Supported git hash: 5c88c5996dbde6208e3bec05abc21ff6cd822d26

rm -rf build

jobla@DESKTOP-FS36I9G:~/esp32/micropython/ports/esp32$ make V=1

** WARNING **

The git hash of ESP IDF does not match the supported version

The build may complete and the firmware may work but it is not guaranteed

ESP IDF path:       /home/jobla/newidf/esp-idf

Current git hash:   5c88c5996dbde6208e3bec05abc21ff6cd822d26

Supported git hash: 5c88c5996dbde6208e3bec05abc21ff6cd822d26

mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/bootloader/subproject/main/

mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/bootloader_support/src/

mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/log/

mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/micro-ecc/micro-ecc/

mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/soc/esp32/

mkdir -p build/bootloader//home/jobla/newidf/esp-idf/components/spi_flash/

CC /home/jobla/newidf/esp-idf/components/soc/esp32/rtc_clk.c

xtensa-esp32-elf-gcc -std=gnu99 -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -I../../lib/mp-readline -I../../lib/netutils -I../../lib/timeutils -Ibuild -I/home/jobla/newidf/esp-idf/components/bootloader_support/include -I/home/jobla/newidf/esp-idf/components/bootloader_support/include_bootloader -I/home/jobla/newidf/esp-idf/components/driver/include -I/home/jobla/newidf/esp-idf/components/driver/include/driver -I/home/jobla/newidf/esp-idf/components/nghttp/port/include -I/home/jobla/newidf/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/jobla/newidf/esp-idf/components/esp32/include -I/home/jobla/newidf/esp-idf/components/soc/include -I/home/jobla/newidf/esp-idf/components/soc/include/soc -I/home/jobla/newidf/esp-idf/components/soc/esp32/include -I/home/jobla/newidf/esp-idf/components/ethernet/include -I/home/jobla/newidf/esp-idf/components/expat/expat/expat/lib -I/home/jobla/newidf/esp-idf/components/expat/port/include -I/home/jobla/newidf/esp-idf/components/heap/include -I/home/jobla/newidf/esp-idf/components/json/include -I/home/jobla/newidf/esp-idf/components/json/port/include -I/home/jobla/newidf/esp-idf/components/log/include -I/home/jobla/newidf/esp-idf/components/newlib/platform_include -I/home/jobla/newidf/esp-idf/components/newlib/include -I/home/jobla/newidf/esp-idf/components/nvs_flash/include -I/home/jobla/newidf/esp-idf/components/freertos/include -I/home/jobla/newidf/esp-idf/components/tcpip_adapter/include -I/home/jobla/newidf/esp-idf/components/lwip/include/lwip -I/home/jobla/newidf/esp-idf/components/lwip/include/lwip/port -I/home/jobla/newidf/esp-idf/components/lwip/include/lwip/posix -I/home/jobla/newidf/esp-idf/components/mbedtls/mbedtls/include -I/home/jobla/newidf/esp-idf/components/mbedtls/port/include -I/home/jobla/newidf/esp-idf/components/spi_flash/include -I/home/jobla/newidf/esp-idf/components/ulp/include -I/home/jobla/newidf/esp-idf/components/vfs/include -I/home/jobla/newidf/esp-idf/components/xtensa-debug-module/include -I/home/jobla/newidf/esp-idf/components/wpa_supplicant/include -I/home/jobla/newidf/esp-idf/components/wpa_supplicant/port/include -I/home/jobla/newidf/esp-idf/components/ethernet/include -I/home/jobla/newidf/esp-idf/components/app_trace/include -I/home/jobla/newidf/esp-idf/components/app_update/include -I/home/jobla/newidf/esp-idf/components/pthread/include -I/home/jobla/newidf/esp-idf/components/smartconfig_ack/include -DIDF_VER=\"v3.3-beta1-268-g5c88c5996\" -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_BTREE=1 -DBOOTLOADER_BUILD=1 -I/home/jobla/newidf/esp-idf/components/bootloader_support/include_priv -I/home/jobla/newidf/esp-idf/components/bootloader_support/include -I/home/jobla/newidf/esp-idf/components/micro-ecc/micro-ecc -I/home/jobla/newidf/esp-idf/components/esp32 -Wno-error=format -c -MD -o build/bootloader//home/jobla/newidf/esp-idf/components/soc/esp32/rtc_clk.o /home/jobla/newidf/esp-idf/components/soc/esp32/rtc_clk.c

CC /home/jobla/newidf/esp-idf/components/soc/esp32/rtc_time.c

xtensa-esp32-elf-gcc -std=gnu99 -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -I../../lib/mp-readline -I../../lib/netutils -I../../lib/timeutils -Ibuild -I/home/jobla/newidf/esp-idf/components/bootloader_support/include -I/home/jobla/newidf/esp-idf/components/bootloader_suppor [... cut ...] ents/heap/include -I/home/jobla/newidf/esp-idf/components/json/include -I/home/jobla/newidf/esp-idf/components/json/port/include -I/home/jobla/newidf/esp-idf/components/log/include -I/home/jobla/newidf/esp-idf/components/newlib/platform_include -I/home/jobla/newidf/esp-idf/components/newlib/include -I/home/jobla/newidf/esp-idf/components/nvs_flash/include -I/home/jobla/newidf/esp-idf/components/freertos/include -I/home/jobla/newidf/esp-idf/components/tcpip_adapter/include -I/home/jobla/newidf/es

dpgeorge, 2019-02-21:
There is some log missing at the end, and it would be helpful to have the last few lines, especially the line just after "LINK ...."

But, from the build log that is given, it looks like make is not building some files that it should be (eg soc/esp32/rtc_wdt.o).  It's very strange indeed.  All I can conclude is that there is something wrong with the system that it's being built on, either the OS, local filesystem, or make is not working properly.

I suggest trying a Linux system for the build, eg in a virtual machine.

johncblacker, 2019-02-21:
Thanks for the response, here's a lower portion of the output with a slight overlap:
------------------------ log paste ---------------------
CC /home/jobla/newidf/esp-idf/components/log/log.c
xtensa-esp32-elf-gcc -std=gnu99 -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -I../../lib/mp-readline -I../../lib/netutils -I../../lib/timeutils -Ibuild -I/home/jobla/newidf/esp-idf/components/bootloader_support/include -I/home/jobla/newidf/esp-idf/components/bootloader_support/include_bootloader -I/home/jobla/newidf/esp-idf/components/driver/include -I/home/jobla/newidf/esp-idf/components/driver/include/driver -I/home/jobla/newidf/esp-idf/components/nghttp/port/include -I/home/jobla/newidf/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/jobla/newidf/esp-idf/components/esp32/include -I/home/jobla/newidf/esp-idf/components/soc/include -I/home/jobla/newidf/esp-idf/components/soc/include/soc -I/home/jobla/newidf/esp-idf/components/soc/esp32/include -I/home/jobla/newidf/esp-idf/components/ethernet/include -I/home/jobla/newidf/esp-idf/components/expat/expat/expat/lib -I/home/jobla/newidf/esp-idf/components/expat/port/include -I/home/jobla/newidf/esp-idf/components/heap/include -I/home/jobla/newidf/esp-idf/components/json/include -I/home/jobla/newidf/esp-idf/components/json/port/include -I/home/jobla/newidf/esp-idf/components/log/include -I/home/jobla/newidf/esp-idf/components/newlib/platform_include -I/home/jobla/newidf/esp-idf/components/newlib/include -I/home/jobla/newidf/esp-idf/components/nvs_flash/include -I/home/jobla/newidf/esp-idf/components/freertos/include -I/home/jobla/newidf/esp-idf/components/tcpip_adapter/include -I/home/jobla/newidf/esp-idf/components/lwip/include/lwip -I/home/jobla/newidf/esp-idf/components/lwip/include/lwip/port -I/home/jobla/newidf/esp-idf/components/lwip/include/lwip/posix -I/home/jobla/newidf/esp-idf/components/mbedtls/mbedtls/include -I/home/jobla/newidf/esp-idf/components/mbedtls/port/include -I/home/jobla/newidf/esp-idf/components/spi_flash/include -I/home/jobla/newidf/esp-idf/components/ulp/include -I/home/jobla/newidf/esp-idf/components/vfs/include -I/home/jobla/newidf/esp-idf/components/xtensa-debug-module/include -I/home/jobla/newidf/esp-idf/components/wpa_supplicant/include -I/home/jobla/newidf/esp-idf/components/wpa_supplicant/port/include -I/home/jobla/newidf/esp-idf/components/ethernet/include -I/home/jobla/newidf/esp-idf/components/app_trace/include -I/home/jobla/newidf/esp-idf/components/app_update/include -I/home/jobla/newidf/esp-idf/components/pthread/include -I/home/jobla/newidf/esp-idf/components/smartconfig_ack/include -DIDF_VER=\"v3.3-beta1-268-g5c88c5996\" -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_BTREE=1 -DBOOTLOADER_BUILD=1 -I/home/jobla/newidf/esp-idf/components/bootloader_support/include_priv -I/home/jobla/newidf/esp-idf/components/bootloader_support/include -I/home/jobla/newidf/esp-idf/components/micro-ecc/micro-ecc -I/home/jobla/newidf/esp-idf/components/esp32 -Wno-error=format -c -MD -o build/bootloader//home/jobla/newidf/esp-idf/components/log/log.o /home/jobla/newidf/esp-idf/components/log/log.c
AR build/bootloader/liblog.a
xtensa-esp32-elf-ar cr build/bootloader/liblog.a build/bootloader//home/jobla/newidf/esp-idf/components/log/log.o
CC /home/jobla/newidf/esp-idf/components/spi_flash/spi_flash_rom_patch.c
xtensa-esp32-elf-gcc -std=gnu99 -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -I../../lib/mp-readline -I../../lib/netutils -I../../lib/timeutils -Ibuild -I/home/jobla/newidf/esp-idf/components/bootloader_support/include -I/home/jobla/newidf/esp-idf/components/bootloader_support/include_bootloader -I/home/jobla/newidf/esp-idf/components/driver/include -I/home/jobla/newidf/esp-idf/components/driver/include/driver -I/home/jobla/newidf/esp-idf/components/nghttp/port/include -I/home/jobla/newidf/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/jobla/newidf/esp-idf/components/esp32/include -I/home/jobla/newidf/esp-idf/components/soc/include -I/home/jobla/newidf/esp-idf/components/soc/include/soc -I/home/jobla/newidf/esp-idf/components/soc/esp32/include -I/home/jobla/newidf/esp-idf/components/ethernet/include -I/home/jobla/newidf/esp-idf/components/expat/expat/expat/lib -I/home/jobla/newidf/esp-idf/components/expat/port/include -I/home/jobla/newidf/esp-idf/components/heap/include -I/home/jobla/newidf/esp-idf/components/json/include -I/home/jobla/newidf/esp-idf/components/json/port/include -I/home/jobla/newidf/esp-idf/components/log/include -I/home/jobla/newidf/esp-idf/components/newlib/platform_include -I/home/jobla/newidf/esp-idf/components/newlib/include -I/home/jobla/newidf/esp-idf/components/nvs_flash/include -I/home/jobla/newidf/esp-idf/components/freertos/include -I/home/jobla/newidf/esp-idf/components/tcpip_adapter/include -I/home/jobla/newidf/esp-idf/components/lwip/include/lwip -I/home/jobla/newidf/esp-idf/c [... cut ...]
build/bootloader/libbootloader_support.a(bootloader_clock.o):(.literal.bootloader_clock_configure+0x14): undefined reference to `rtc_clk_init'
build/bootloader/libbootloader_support.a(bootloader_clock.o): In function `bootloader_clock_configure':
bootloader_clock.c:(.text.bootloader_clock_configure+0x76): undefined reference to `rtc_clk_init'
collect2: error: ld returned 1 exit status
Makefile:813: recipe for target 'build/bootloader.elf' failed
make: *** [build/bootloader.elf] Error 1
---------------------- end of pasted output -----------------------------
Will make an attempt to build on my linux ubuntu standalone system today.

johncblacker, 2019-02-21:
I've tried building on my standalone ubuntu 16.04 LTS system and got the following error:
----------------------------------- build output ------------------------------------------------
LINK build/bootloader.elf
xtensa-esp32-elf-gcc  -nostdlib -L/home/john/esp-idf/lib -L/home/john/esp-idf/ld -u call_user_start_cpu0 -Wl,--gc-sections -static -Wl,-EL -Wl,-Map=build/bootloader.map -Wl,--cref -T /home/john/esp-idf/components/bootloader/subproject/main/esp32.bootloader.ld -T /home/john/esp-idf/components/bootloader/subproject/main/esp32.bootloader.rom.ld -T /home/john/esp-idf/components/esp32/ld/esp32.rom.ld -T /home/john/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld -T /home/john/esp-idf/components/esp32/ld/esp32.peripherals.ld -o build/bootloader.elf  -Wl,--start-group build/bootloader//home/john/esp-idf/components/soc/esp32/rtc_clk.o build/bootloader//home/john/esp-idf/components/soc/esp32/rtc_clk_init.o build/bootloader//home/john/esp-idf/components/soc/esp32/rtc_init.o build/bootloader//home/john/esp-idf/components/soc/esp32/rtc_time.o build/bootloader//home/john/esp-idf/components/soc/esp32/rtc_wdt.o build/bootloader//home/john/esp-idf/components/soc/esp32/cpu_util.o build/bootloader//home/john/esp-idf/components/bootloader/subproject/main/bootloader_start.o -Lbuild/bootloader -lbootloader_support -llog -lspi_flash -lmicro-ecc -L/home/john/esp-idf/components/esp32/lib -lrtc -L/home/john/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/ -lgcc -Wl,--end-group
xtensa-esp32-elf-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found
compilation terminated.
Makefile:974: recipe for target 'build/bootloader.elf' failed
make: *** [build/bootloader.elf] Error 1
---------------------------------------------------- end of output paste -------------------------------
don't think I missed anything, did git submodule update --recursive on both my MP repository and my esp-idf repository, did a git checkout 5c88c5996dbde6208e3bec05abc21ff6cd822d26 which is the hash of the esp-idf in the esp32 port Makefile.  Also, I believe I've installed the requisite dependencies, but apparently something is missing?

johncblacker, 2019-02-21:
I decided to start from scratch. But this time I put the micropython repository inside xtensa-esp32-elf directory and got a lot further; however, now I'm getting a hangup saying pyparsing module not found; however, I do a pip list and pyparsing shows, also I get into the python interpreter and issue an import for pyparsing and it works fine.  Here's the error:
------------------------- build output error paste ------------------------------
jobla@DESKTOP-FS36I9G:~/esp32/xtensa-esp32-elf/micropython/ports/esp32$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/esp32.common.ld
Traceback (most recent call last):
  File "/home/jobla/newidf/esp-idf/tools/ldgen/ldgen.py", line 22, in <module>
    from fragments import FragmentFileModel
  File "/home/jobla/newidf/esp-idf/tools/ldgen/fragments.py", line 20, in <module>
    from sdkconfig import SDKConfig
  File "/home/jobla/newidf/esp-idf/tools/ldgen/sdkconfig.py", line 18, in <module>
    from pyparsing import Word, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
ModuleNotFoundError: No module named 'pyparsing'
Makefile:780: recipe for target 'build/esp32.common.ld' failed
make: *** [build/esp32.common.ld] Error 1
jobla@DESKTOP-FS36I9G:~/esp32/xtensa-esp32-elf/micropython/ports/esp32$ python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34)
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyparsing
>>> quit()
jobla@DESKTOP-FS36I9G:~/esp32/xtensa-esp32-elf/micropython/ports/esp32$ python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34)
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyparsing import Word, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
>>> quit()
------------------------- end of paste -------------------------------------------------
I also included the command line executions of python importing pyparsing.

johncblacker, 2019-02-21:
I'm sorry, the previous comment paste didn't work right:
------------------------ build output -----------------------------------------
jobla@DESKTOP-FS36I9G:~/esp32/xtensa-esp32-elf/micropython/ports/esp32$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/esp32.common.ld
Traceback (most recent call last):
  File "/home/jobla/newidf/esp-idf/tools/ldgen/ldgen.py", line 22, in <module>
    from fragments import FragmentFileModel
  File "/home/jobla/newidf/esp-idf/tools/ldgen/fragments.py", line 20, in <module>
    from sdkconfig import SDKConfig
  File "/home/jobla/newidf/esp-idf/tools/ldgen/sdkconfig.py", line 18, in <module>
    from pyparsing import Word, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
ModuleNotFoundError: No module named 'pyparsing'
Makefile:780: recipe for target 'build/esp32.common.ld' failed
make: *** [build/esp32.common.ld] Error 1
jobla@DESKTOP-FS36I9G:~/esp32/xtensa-esp32-elf/micropython/ports/esp32$ python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34)
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyparsing
>>> quit()
jobla@DESKTOP-FS36I9G:~/esp32/xtensa-esp32-elf/micropython/ports/esp32$ python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34)
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyparsing import Word, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
>>> quit()
----------------------------------  end of paste ---------------------------------

johncblacker, 2019-02-21:
I don't know what's screwing up the paste...but you can see the error behind the strikeouts.
jb

johncblacker, 2019-02-21:
I took a stab at this again and found I had both python 2.7 and python 3 installed.  I was thinking perhaps the version of python that was being used was 3, so I did a pip3 list and pyparsing wasn't there.  So, I installed via pip3 install pyparsing; pip3 install blockstack and then did my make again and it was 
SUCCESSFUL!  Wrote out a firmware.bin.  So, I'm going to close this issue, but thank you for your attention to the issue.

closed 2019-02-21 by johncblacker