micropython/micropython #4524: could not import ubluepy module in nrf52832

rlnktt, 2019-02-19:
Hello everyone !! 

I have built the firmware 1.10 for nrf port successfully. After flashing the chip with the firmware, I couldn't import the ubluepy module, which I need to run some examples,  given in examples directory.

I am getting the below error

[image]

Any help is greatly appreciated

glennrub, 2019-02-20:
@rlnktt , are you sure that you flashed using "make <board> sd" which also flash the SoftDevice?
I've seen this myself when i forget to flash the BLE stack, ending up in running the old firmware starting from address 0x0 in ROM, which is compiled without BLE support.

rlnktt, 2019-02-20:
Thanks @glennrub . Inclusion of SD=132 sd in the make command , fixed it. But the device doesn't seem to show up REPL UART now!!

glennrub, 2019-02-20:
@rlnktt , could sound like you have enabled MICROPY_PY_BLE_NUS in bluetooth_config.h, redirecting UART to BLE UART. However, i believe we should should move this discussion to the forum. I would be more than happy to help you get booted up on the nrf52832 there. It will also help new users later.
https://forum.micropython.org/viewforum.php?f=12

rlnktt, 2019-03-08:
@glennrub , Actually I have not enabled it in the so said configuration. As per your suggestion, I have moved this to MicroPython Forum

See here : https://forum.micropython.org/viewtopic.php?f=12&t=6061

jonnor, 2024-07-27:
Note that there is now work to replace this with the `bluetooth` module used on other ports. Ref https://github.com/micropython/micropython/pull/8318

closed 2024-12-27 by jonnor