micropython/micropython #4527: Generator `.throw(e, None)` and `.throw(e, None, None)` mishandle already-instantiated exception

mentalisttraceur, 2019-02-20:
Using [the web browser Unicorn emulator](https://micropython.org/unicorn/), haven't verified against newer versions:

```python
>>> def g(): yield
...
>>> g().throw(Exception('koz'), None)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 1, in g
TypeError: exceptions must derive from BaseException
```

The expected (what CPython does) behavior is for it to just raise `Exception('koz')`, same as it would if we didn't pass the second argument at all. Same with when all three arguments are passed in.

The CPython logic for determining when the first argument is allowed to be an exception seems to be based on the second argument **either** being not passed in **or** `None`.

This is not a significant issue and the workaround in calling code is short and simple, but it can cause unexpected and not-obvious breakages deep inside code that otherwise works, and I presume the fix would be trivial for someone who knows where and how `.throw` is implemented in MicroPython, and I presume it would have almost no performance impact (one conditional inside just the code path of raising exceptions in a generator).

dpgeorge, 2019-02-21:
Thanks for the report.  Indeed it's quite minor, and the fix is also minor, see #4534

pfalcon, 2019-02-23:
@mentalisttraceur: Can you comment on where the code which causes this issue comes from? (So we better understand what usecase we're fixing).

mentalisttraceur, 2019-02-24:
@dpgeorge Thank you for handling it so quickly and readily!

@pfalcon The fix by @dpgeorge fixes the usecase I care about. Were you looking for a more concrete or motivated example?

pfalcon, 2019-02-24:
@mentalisttraceur: Yes, I'm asking where that usecase comes from. Did you it yourself? Does it come from some specific software?

dpgeorge, 2019-05-09:
Should be fixed by dac9d4767175541aaaaf0e2f873322f5f1db3b0c

closed 2019-05-09 by dpgeorge