micropython/micropython #4529: "-fm dout" flashing option to BlitzWolf SHP6

robertio, 2019-02-20:
Hi 
I figured out a working flashing option for BlitzWolf SHP6 (which is yet another smart socket.)
see picture how to serial connect [url]https://github.com/arendst/Sonoff-Tasmota/wiki/BlitzWolf-SHP6[/url]

[code]esptool.py --port /dev/ttyUSB0 erase_flash 
esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_size=detect  -fm dout 0 esp8266-20190125-v1.10.bin
[/code]

after flash you can connect to serial terminal.

dpgeorge, 2019-02-21:
Thanks for the info, and I see you've already shared it on the forum which is the right place for this.  So closing this issue.

closed 2019-02-21 by dpgeorge

robertio, 2019-02-21:
Hi George!

Many thanks for highly organised documentation at docs.micropython.org! :)

Do you think it would be useful to mention "dout" in section besides "dio" option? ( I spent it a few hours to figure out)
https://docs.micropython.org/en/latest/esp8266/tutorial/intro.html#intro

"For some boards with a particular FlashROM configuration (e.g. some variants of a NodeMCU board) you may need to use the following command to deploy the firmware (note the -fm dio (or dout) option):"

Best regards
Robert