micropython/micropython #4536: Update .mpy version to version 4

dpgeorge, 2019-02-21:
The following PRs require a new .mpy file format version so it would be good to merge them around the same time so the .mpy version only needs to be updated once:
- py/persistentcode: Only store a single instance of source_file in mpy. #4472
- py: Define a set of static qstrs to reduce size of mpy files. #4473
- py: Fix unwinding of try-finally with break in finally clause #4504
- Add support to save native, viper and inline-asm code to .mpy files #4535

peterhinch, 2019-02-21:
This all sounds awesome :)

I have an ESP8266 application intended to run under the current release build 1.10. I distribute a .mpy file for users unable to do the cross-compilation. If mpy-cross changes to produce a new format, will there be a way to create mpy files compatible with V1.10? Other than the obvious one of keeping a copy of the current version.

Ideally mpy-cross would have a way of specifying specific versions.

hoihu, 2019-02-21:
yes that sounds really exiting!

we are using mpy files a lot and using native / viper emitters within mpy is a big thing! Really looking forward to this.

perhaps yet another step is to support those emitters within frozen bytecode. 😉

dpgeorge, 2019-02-22:
> If mpy-cross changes to produce a new format, will there be a way to create mpy files compatible with V1.10?

From the point of view of supporting the layout/format of the .mpy file, yes that would be possible (eg by using the old version 3 of .mpy for code that didn't contain any native blocks).  But the issue is that the bytecode itself changed in #4504, so the output of the bytecode compiler is not compatible with the VM, for different mpy versions.  So really the only way to do it is generate separate .mpy files for v1.10 and for the next release, using two different mpy-cross builds.

> perhaps yet another step is to support those emitters within frozen bytecode

Yes that's definitely the next step :)

pfalcon, 2019-02-26:
> so it would be good to merge them around the same time so the .mpy version only needs to be updated once:

Incrementing the version once for all those changes sounds good, but piling them up doesn't. At the very least, those are all together massive changes, they may contain bugs, and merging them as a big code drop may make it hard to figure out what caused what. Folks working with master should be prepared to rebuild mpy-cross and bytecode with each work session anyway. So, starting to merge ready PRs and have sustained development is a good idea.

andrewleech, 2019-03-04:
@peterhinch if it helps I've got an auto-builder set up pumping out releases of mpy-cross to pypi for convenient install/use: https://pypi.org/project/mpy-cross/ https://gitlab.com/alelec/mpy_cross 
If you even need an older version of the tool to go along with older releases, you can pip install an older copy for most platforms pretty easily from there.

dpgeorge, 2019-03-08:
The mpy version was updated in 9a5f92ea72754c01cc03e5efcdfe94021120531e

closed 2019-03-08 by dpgeorge