micropython/micropython #4540: RTC cannot be set correctly on ESP 32

nedoskiv, 2019-02-22:
Hello, 
I got troubles using time/clock on ESP32, apart that internal clock (or whatever it have) speed up with around 80% when i tried to set it, it do not set correctly:

>>> rtc = machine.RTC()
>>> rtc.init((2019, 2, 22, 11, 40, 0, 0, 0))
>>> rtc.datetime()
(2019, 2, 23, 5, 16, 0, 7, 558758)

try on old and latest release it is the same (MicroPython v1.10-98-g4daee3170)

mrwhy-orig, 2019-02-25:
I guess it is the same issue like this: #3674
The documentation is wrong here. Parameter 4 is the day of week and gets always corrected no matter whats inside. 5 means it is saturday.

nedoskiv, 2019-02-25:
I give up with trying to set proper time, so my log files timestamp write  seconds after restart.

GHPS, 2019-04-11:
Please close this issue - it is solved or - precisely - was never a problem in MicroPython.

nedoskiv, 2019-04-11:
Well I do not tried this on latest releases, but gonna close it, and reopen again if face same problem

closed 2019-04-11 by nedoskiv

PrimeTime416, 2020-02-01:
I do not agree there was never a problem, the OG documentation was not clear, hopefully the configuration information and example below helps clarify how to setting the ESP32 RTC with MicroPython.

The 8-tuple has the following format:
(year, month, day, weekday, hours, minutes, seconds, sub-seconds)

(0) year includes the century (for example 2020).
(1) month is 1-12
(2) mday is 1-31 or yday 1-365/366
(3) weekday is 0-6 for Mon-Sun ***Set to 0, being Auto Set by machine
(4) hour is 0-23
(5) minute is 0-59
(6) second is 0-59
(7) sub-seconds

`import machine

rtc = machine.RTC()
datetime = rtc.datetime
year = 2019
month = 12
day = 31
_weekday = 0
hours = 19
minutes = 05
seconds = 0
subseconds = 0

datetime_format = "{2}, {1} {3}, {0}\nTime: {4}:{5}:{6}.{7}"

def get_time_date(getX, valueX):
    weekday_list = ('Monday', 'Tuesday', 'Wednesday', 'Thursday', 'Friday', 'Saturday', 'Sunday')
    month_list = ('January', 'February', 'March', 'April', 'May', 'June', 'July', 'August', 'September', 'Octber', 'December', 'January')
    if getX == "weekday" : return weekday_list[valueX]
    elif getX == "month" : return month_list[valueX - 1]

rtc.init((year, month, day, _weekday, hours, minutes, seconds, subseconds))

print(datetime_format.format(
    datetime()[0], 
    get_time_date('month', datetime()[1]), 
    get_time_date('weekday', datetime()[3]), 
    datetime()[2], 
    datetime()[4], 
    datetime()[5],
    datetime()[6],
    datetime()[7],
))`

OUTPUT:
Tuesday, January 31, 2019
Time: 19:5:0.1325

azwar, 2021-02-14:
> month_list = ('January', 'February', 'March', 'April', 'May', 'June', 'July', 'August', 'September', 'Octber', 'December', 'January')

Hi @PrimeTime416 thank you for giving this sample code. I tried it and the output is just as the same you mention. After reviewing your sample, you have mistake at **month_list** part. You did not put November and also you have double January. There are January at first index and last index. 

The variable of month_list should be:
`month_list = ('January', 'February', 'March', 'April', 'May', 'June', 'July', 'August', 'September', 'Octber', 'November', 'December')`