micropython/micropython #4542: !!!!!compile error !!!

c-hamster, 2019-02-23:
Hi ,I download the source code from github,and build it as what README.md.but It doesn't work.

.....
CC ../../extmod/vfs_fat_diskio.c
CC ../../extmod/vfs_fat_file.c
CC ../../extmod/utime_mphal.c
CC ../../extmod/uos_dupterm.c
CC ../../lib/embed/abort_.c
CC ../../lib/utils/printf.c
GEN build-PYBV10/frozen_mpy.c
**usage: mpy-tool.py [-h] [-d] [-f] [-q QSTR_HEADER]**
                   **[-mlongint-impl {none,longlong,mpz}] [-mmpz-dig-size N]
                   files [files ...]
mpy-tool.py: error: the following arguments are required: files
../../py/mkrules.mk:122: recipe for target 'build-PYBV10/frozen_mpy.c' failed**
make: *** [build-PYBV10/frozen_mpy.c] Error 2
make: *** Deleting file 'build-PYBV10/frozen_mpy.c'
python@linuxpc:/mnt/hgfs/wshare/micropython/ports/stm32$

dpgeorge, 2019-02-25:
It's possible symlinks don't work on your system.  Try rebuilding (after make clean) with:
```
$ make FROZEN_MPY_DIR=
```

dpgeorge, 2019-03-26:
I assume this was fixed by the above suggestion.  If not feel free to reopen with further details.

closed 2019-03-26 by dpgeorge