micropython/micropython #4546: `dir()` executes properties. It shouldn't.

smurfix, 2019-02-24:
`dir(OBJ)` enumerates the attributes of an object. It should not call its properties. That's inefficent, unexpected, and a deviation from CPython.

This is a problem for me because I am doing a minimal reimplementation of `attrs`, which needs an object's directory while its setup hasn't completed yet.

```
>>> class x:
...     @property
...     def y(self):
...         raise ValueError("not yet")
... 
>>> xx=x()
>>> dir(xx)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 4, in y
ValueError: not yet
```

dpgeorge, 2019-02-26:
This behaviour was introduced in 98647e83c733a8a8d36522f71e677367ca8ddd03, and see #3617 for discussion about it.

Since `dir()` (and tab completion, it works the same) are mainly for interactive use it was deemed more important to make them work well rather than efficiently.  And by "work well" it means to list items from parent classes, handling multiple inheritance and without duplication.

Maybe it's possible to find another way to implement `dir()` so it is more efficient and has less side effects.

In the meantime, a workaround is to implement `__dir__`:
```python
class x:
    def __dir__(self):
        return []
    ...
```

smurfix, 2019-02-26:
Bah. Thank you, I'll have to think of a more efficient solution for my usecase.

dpgeorge, 2019-02-26:
Let's keep this open because probably it should be improved.