micropython/micropython #4550: Can't execute files from FS using pyexec_file

orpheeantoniadis, 2019-02-25:
Hello,

I am currently porting micropython for a project using an LPC1759. I want to have a FAT file system and execute files from the C code and not from the micropython REPL.

I tried to use the function `pyexec_file` but it didn't work and I cannot find any documentation about how to use it. I understand that I must implement `mp_import_stat` and `mp_lexer_new_from_file` but I don't know what `pyexec_file` is expecting those functions to do.

I use oofatfs lib for the FAT implementation and everything is working well. I can initialize the FS in the flash of my microcontroller and create some files. I can see the files when I mount the flash as a USB mass storage device. I can also execute the content of the file when I import it directly in the REPL using a serial interface so it is not a problem with the python code. By the way, the code is really simple, it just turns on some leds.

If someone could help me it would be great, I am looking for a solution for 2 days now and nothing is working.

Regards,

Orphee Antoniadis

dpgeorge, 2019-02-25:
> I tried to use the function `pyexec_file` but it didn't work

Please describe more about how it didn't work, what happened and what you expected to happen, etc.  Also describe where you called it from and how.

orpheeantoniadis, 2019-02-25:
I use it like so (I took for example the stm32 port) :

```
const char *test_py = "test.py";
mp_import_stat_t stat = mp_import_stat(test_py);
if (stat == MP_IMPORT_STAT_FILE) {
	int ret = pyexec_file(test_py);
}
```

I am expecting the script test.py running. It turns on some leds but nothing is happening.
I send you also my vfs fat lexer :

```
typedef struct _mp_lexer_file_buf_t {
    FIL fp;
    byte buf[20];
    uint16_t len;
    uint16_t pos;
} mp_lexer_file_buf_t;

STATIC mp_uint_t file_buf_next_byte(mp_lexer_file_buf_t *fb) {

    if (fb->pos >= fb->len) {
        if (fb->len < sizeof(fb->buf)) {
            return MP_LEXER_EOF;
        } else {
            UINT n;
            f_read(&fb->fp, fb->buf, sizeof(fb->buf), &n);
            if (n == 0) {
                return MP_LEXER_EOF;
            }
            fb->len = n;
            fb->pos = 0;
        }
    }
    return fb->buf[fb->pos++];
}

STATIC void file_buf_close(mp_lexer_file_buf_t *fb) {
    f_close(&fb->fp);
    m_del_obj(mp_lexer_file_buf_t, fb);
}

mp_lexer_t *fat_vfs_lexer_new_from_file(const char *filename) {
    mp_lexer_file_buf_t *fb = m_new_obj_maybe(mp_lexer_file_buf_t);
    if (fb == NULL) {
        return NULL;
    }
    FRESULT res = f_open(&fs, &fb->fp, filename, FA_READ);
    if (res != FR_OK) {
        m_del_obj(mp_lexer_file_buf_t, fb);
        return NULL;
    }
    UINT n;
    f_read(&fb->fp, fb->buf, sizeof(fb->buf), &n);
    fb->len = n;
    fb->pos = 0;

    mp_reader_t reader = {fb, (void *)file_buf_next_byte, (void *)file_buf_close};

    return mp_lexer_new(qstr_from_str(filename), reader);
}

mp_import_stat_t mp_import_stat(const char *path) {
	FILINFO fno;
	FRESULT res = f_stat(&fs, path, &fno);
	if (res == FR_OK) {
		if ((fno.fattrib & AM_DIR) != 0) {
			return MP_IMPORT_STAT_DIR;
		} else {
			return MP_IMPORT_STAT_FILE;
		}
	}
	return MP_IMPORT_STAT_NO_EXIST;
}

mp_lexer_t *mp_lexer_new_from_file(const char *filename) {
    return fat_vfs_lexer_new_from_file(filename);
}
```

Thank you.

dpgeorge, 2019-02-26:
The code looks ok, but it's not really possible to debug it just by reading it out of context.  You will need to trace through the execution (eg using printf's) to see what is being called and where the error is.

orpheeantoniadis, 2019-02-26:
Ok so I finally manage to run the file. I just changed the order I call my init functions.

I change my code from below :

```
init_file_system(&fs);
device = new USBSerial(0x1f00, 0x2012, 0x0001, false);
mp_reset();
gc_init(heap, heap + sizeof(heap));
mp_init();

const char *test_py = "test.py";
mp_import_stat_t stat = mp_import_stat(test_py);
if (stat == MP_IMPORT_STAT_FILE) {
	int ret = pyexec_file(test_py);
}
```

To this code :

```
gc_init(heap, heap + sizeof(heap));
init_file_system(&fs);
device = new USBSerial(0x1f00, 0x2012, 0x0001, false);
mp_reset();
mp_init();

const char *test_py = "test.py";
mp_import_stat_t stat = mp_import_stat(test_py);
if (stat == MP_IMPORT_STAT_FILE) {
	int ret = pyexec_file(test_py);
}
```

The only difference is the call of the `gc_init` function. It's a bit weird because the `init_file_system` function just creates the FS and the files and doesn't malloc anything. Like I said in a previous message, there was no bug with this function before so I don't know why it is working now. Maybe you have an idea why.

Anyway, thank you for your answers, I think you can close the issue now.

dpgeorge, 2019-02-26:
I suggest you try running the whole MicroPython test suite to see if there are any issues with memory allocation, or anything else.  It might point to some areas of the port that can be improved.

closed 2019-02-26 by dpgeorge