micropython/micropython #4552: Module dictionary corruption

smurfix, 2019-02-25:
The problem: I'm trying to port https://github.com/python-trio/trio to micropython. During this I've hit a problem with apparent module globals corruption that has me stumped.

Specifically, as part of that work I have backported the `contextvars` module to micropython-lib. This module is imported correctly. It has a nice global `_set_context` function … which happens to suddenly not be there any more:
```
  File "/srv/src/contextvars/contextvars/__init__.py", line 28, in run
NameError: name '_set_context' isn't defined
```

## How to reproduce

You need `http://github.com/smurfix/micropython-lib`, `master` branch, and `http://github.com/smurfix/trio`, `micro` branch, and this test program:
```
#!./micropython

import sys
sys.path[0] = "/tmp/lib" # wherever you installed micropython-lib to
sys.path.insert(0,"/srv/src/trio")  # wherever you checked out trio
import trio

async def run():
	async with trio.open_nursery() as n:
		await trio.sleep(1)
		print("Yes")

trio.run(run)
```

Running this with (the `master` branch of the Unix port of) micropython results in this stack trace:
```
Traceback (most recent call last):
  File "t.mpy", line 20, in <module>
  File "/srv/src/trio/trio/_core/_run.py", line 1383, in run
  File "/srv/src/trio/trio/_core/_run.py", line 1371, in run
  File "/srv/src/trio/trio/_core/_run.py", line 1369, in run
  File "/srv/src/trio/trio/_core/_run.py", line 1369, in run
  File "/srv/src/trio/trio/_core/_run.py", line 1365, in run
  File "/srv/src/trio/trio/_core/_run.py", line 1365, in run
  File "/srv/src/trio/trio/_core/_run.py", line 1504, in run_impl
  File "/srv/src/trio/trio/_core/_run.py", line 1499, in run_impl
  File "/srv/src/trio/trio/_core/_run.py", line 1495, in run_impl
  File "/srv/src/contextvars/contextvars/__init__.py", line 28, in run
NameError: name '_set_context' isn't defined
```
despite there being a perfectly valid `def _set_context()` declaration in that file. The numerous tests for `contextvars` pass, so there seems to be something more sinister going on.

Needless to say this is quite difficult to reduce to a more self-contained subset that still shows the problem. Ideas how to figure this out would be appreciated.

dpgeorge, 2019-02-26:
Thanks very much for the report, and also steps to reproduce.  I managed to reproduce it, and it's a stack underflow in the VM coming from the mishandling of unwinding an async with statement as an exception is raised through its `__aexit__` handling code.

The issue is that the async with code checks if an object is an exception by testing if it's derived from `Exception`, rather than from `BaseException`.  This code was introduced in d8dc918deb8d4b13b8919706f9f208542c9ef2e6, although that commit fixed other problems with async with.

The reason the bug was hit is because the user-defined `Cancelled` exception (in trio) is derived from `BaseException`.

The fix (apart from making `Cancelled` derive from `Exception`) is to check in the compilation of async-with that objects derive from `BaseException` rather than `Exception`.

A small test that exhibits the bug is:
```python
class AContext:
    async def __aenter__(self):
        print('enter')
        return 1
    async def __aexit__(self, exc_type, exc, tb):
        print('exit', exc_type, exc)

async def f():
    async with AContext():
        print('body')
        raise BaseException # or KeyboardInterrupt or SystemExit

o = f()
try:
    o.send(None)
except BaseException:
    print('finished')
```
The `__aexit__` method won't be called, but it should be.

dpgeorge, 2019-02-26:
See #4554 for a fix.

dpgeorge, 2019-03-05:
The fix was merged in 12ce9f268909435b49f558d1d3c15ad7591d6fe8

closed 2019-03-05 by dpgeorge