micropython/micropython #4558: Docs don't link to Pyboard v1.1 schematic

johnthagen, 2019-02-26:
See https://docs.micropython.org/en/latest/pyboard/general.html#the-pyboard-hardware

Only the v1.0 schematics are linked, but all new purchases will be v1.1.

johnthagen, 2019-02-27:
Here is a link to the v1.1 schematic: https://micropython.org/resources/PYBv11-schematics.pdf

johnthagen, 2019-02-27:
Also, should this page default to the v1.1 pinout? 

https://docs.micropython.org/en/latest/pyboard/quickref.html

On this page, the v1.1 pinout is at least linked, so it's not as big of a deal.

johnthagen, 2019-03-06:
I also noticed that the v1.1 schematic does not include the layout (which is present in the v1.0). Is there a combined PDF that has all of the information for the (current) v1.1 boards that are being sold?

johnthagen, 2019-03-06:
Related: https://github.com/micropython/pyboard doesn't seem to have been updated in many years. Are there plans to get v1.1 eagle / gerber files up so that all of this information could be linked and made available?

- https://github.com/micropython/pyboard/issues/11

dpgeorge, 2019-03-08:
Thanks for raising this issue.

> I also noticed that the v1.1 schematic does not include the layout (which is present in the v1.0). Is there a combined PDF that has all of the information for the (current) v1.1 boards that are being sold?

I have just uploaded this PDF, it's available as https://micropython.org/resources/PYBv11.pdf

> Also, should this page [quickref] default to the v1.1 pinout?

Yes that makes sense!

> Are there plans to get v1.1 eagle / gerber files up

Not sure about that one, I will see...

johnthagen, 2019-03-08:
@dpgeorge Thanks! Fixed in #4582 now.

dpgeorge, 2019-03-26:
Should now be done.

closed 2019-03-26 by dpgeorge