micropython/micropython #4559: STM32: macro hijacks "free" name

amirgon, 2019-02-26:
On [stm32/mpconfigport.h](https://github.com/micropython/micropython/blob/12ce9f268909435b49f558d1d3c15ad7591d6fe8/ports/stm32/mpconfigport.h#L373):

```c
#define malloc(n) m_malloc(n)
#define free(p) m_free(p)
#define realloc(p, n) m_realloc(p, n)
```

`malloc`, `free` and `realloc` are **defined as macros** here, so they cannot be used elsewhere in a different context.  

For example,

Suppose I have a C struct with a member called `free` which is a function pointer:   
```c
typedef struct {
  int (*free)();
} S;
```
Now if I try to access my member: `s->free()`, **free macro will be invoked**, resulting an error.

**This is not a theoretical issue** - see littlevgl/lv_binding_micropython#11

Possible solution - instead of a macro, define `malloc`, `free` and `realloc` as static inline functions that call `m_malloc`, `m_free` and `m_realloc`.

dpgeorge, 2019-02-26:
Thanks for the report.  It might be worth even getting rid of these since other ports don't provide such macros.

In the meantime, a workaround is to #undef the macros before they are used in different contexts.

dpgeorge, 2019-03-26:
Removed in 781947afdc11887f30957dccc37593caff97b79f

closed 2019-03-26 by dpgeorge