micropython/micropython #4560: Unix __setattr__ and __delattr__ not called

peterhinch, 2019-02-27:
See [this forum post](https://forum.micropython.org/viewtopic.php?f=2&t=6008). The following, adapted from the test script:
```python
class B:
    def __init__(self, d):
        # store the dict in the class, not instance, so
        # we don't get infinite recursion in __getattr_
        B.d = d

    def __getattr__(self, attr):
        if attr in B.d:
            print('__getattr__ {} is {}'.format(attr,B.d))
            return B.d[attr]
        else:
            raise AttributeError(attr)

    def __setattr__(self, attr, value):
        print('__setattr__ {} is {}'.format(attr,B.d))
        B.d[attr] = value

    def __delattr__(self, attr):
        print('__delattr__ {} is {}'.format(attr,B.d))
        del B.d[attr]
        print('after __delattr__ {} is {}'.format(attr,B.d))

a = B({"a":1, "b":2})
# The next statement works in CPython and MicroPython. Prints
# __getattr__ a is {'a': 1, 'b': 2}
# 1
print(a.a)

# But here it goes adrift: MicroPython doesn't call __setattr__
a.b = 5
print(a.b)
print(a.__dict__)

del(a.b)
print(a.b)
```
produces this result under CPython:
```python
__getattr__ a is {'a': 1, 'b': 2}
1
__setattr__ b is {'a': 1, 'b': 2}
__getattr__ b is {'a': 1, 'b': 5}
5
{}
__delattr__ b is {'a': 1, 'b': 5}
after __delattr__ b is {'a': 1}
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/adminpete/rats2.py", line 35, in <module>
    print(a.b)
  File "/home/adminpete/rats2.py", line 12, in __getattr__
    raise AttributeError(attr)
AttributeError: b
```
but this under MicroPython
```python
__getattr__ a is {'a': 1, 'b': 2}
1
5
{'b': 5}
__getattr__ b is {'a': 1, 'b': 2}
2
```
The test script tests/basics/class_delattr_setattr.py doesn't pick this up.

dpgeorge, 2019-02-27:
Thanks for the report.  But I think it's simply that setattr/delattr support is not enabled on the unix port.  But it is for the coverage build, so try testing it with that (make coverage; ./micropython_coverage).

pfalcon, 2019-02-27:
It is also enabled in micropython-dev: https://github.com/micropython/micropython/pull/4309

peterhinch, 2019-03-05:
Apologies - the coverage build works fine.

closed 2019-03-05 by peterhinch