micropython/micropython #4565: Master fails to build for nrf52840

aykevl, 2019-03-01:
When I try to build the current master, this fails:

```
$ make sd BOARD=pca10040 SD=s132
[...]
CC modules/machine/pin.c
modules/machine/pin.c: In function 'pin_print':
modules/machine/pin.c:219:13: error: implicit declaration of function 'nrf_gpio_pin_pull_get' [-Werror=implicit-function-declaration]
     switch (nrf_gpio_pin_pull_get(self->pin)) {
             ^
cc1: all warnings being treated as errors
../../py/mkrules.mk:47: recipe for target 'build-pca10040-s132/modules/machine/pin.o' failed
make: *** [build-pca10040-s132/modules/machine/pin.o] Error 1
```

I haven't yet looked why this is the case.

aykevl, 2019-03-01:
After bisecting this is the commit to blame: e8678cd1c9e36f8a1175773b97ef2df5e5f8b673
This commit was made by me. I'm guessing it broke because of the time between making and merging the PR.

aykevl, 2019-03-01:
Nevermind, it was a matter of updating the submodules.

    git submodule update --init

closed 2019-03-01 by aykevl

drewwestrick, 2019-03-11:
@aykevl I'm getting the same error right now as well. Where did you run that git submodule update command from?

aykevl, 2019-03-11:
In the root directory of esp-idf. Do `echo $ESPIDF` to find it.