micropython/micropython #4566: More modern webrepl interface

joelhoro, 2019-03-01:
The webrepl is great to get started and the websocket implementation seem like a reasonable foundation, but I think the GUI could be enhanced without much (if any) alteration to the server.

I am thinking of 
- ability to view file hierarchy / add folders / files
- simple text editor allowing editing particular files
- scratch pad allowing sending multi-line code to the REPL in one go
- buttons with predefined snippets

All of the enhancements I have in mind seem quite straightforward as they should mostly be done in javascript. They should be doable with the current client/server model.

joelhoro, 2019-03-02:
This is what I have in mind - I am facing some issues w.r.t. communicating with the server, but so far things have been pretty straightforward to implement

[image]

joelhoro, 2019-03-03:
Further progress - would love to get some feedback from people who would use a tool like this

[image]

pfalcon, 2019-03-03:
@joelhoro: As a quick reply. Thanks for the post. Unless you hear something else from the maintainer(s), I'd suggest to treat that as you own separate project, to maintain, promote, support, etc. You posted on the forum, that's right step.

I'd draw your attention to the fact that by now, there're ~ a couple of dozens of various "IDEs" for MicroPython, among them there're also web-based IDEs.

joelhoro, 2019-03-04:
Thanks for that and yeah - it did look like the project was abandoned.

I could not find a decent IDE tbh. 

uPyCraft is the best and it is really pretty flimsy (mine would be better if it were not for the fact that it is not working on serial, but that should be fixable)
pyCharm has a plugin which did not recognize my esp.

I'll look more into the forum threads about micropython IDEs, maybe I'll get some luck.

This being said, I had fun and learned a bunch playing around for a few evenings to where I got :)

hcoohb, 2019-03-24:
@joelhoro Love the new interface with new functionality. It feels a lot more mature than the current one. Way to go!
I hope that micropython developers will take onboard some/all of the feature you suggest. 
As I disagree, it is not because there are other options out there that the official version should be half-baked and missing functionality when people are offering to help... (I do understand though that it takes time to ensure code robustness and sustainability)