micropython/micropython #4574: NRF port - I2C

noblerabbit, 2019-03-04:
Hi,

Did anyone test I2C on the NRF port?

I am using NRF52 DK. I connect the SDA and SCL to 26 and 27 pin on the board.

When I run i2c.scan(), I get [] - an empty list. If I connect the same sensor to esp32 (also running micropython), i2c works as it is supposed to, and I can communicate with the sensor.

Is there anything else I should set on NRF52 DK, besides just connecting the SDA, SCL, and GND and VDD from the sensor to the board; and assigning sda and scl pins through machine.i2c?

```
>>> import machine
>>> i2c = machine.I2C(scl=machine.Pin(27), sda=machine.Pin(26))
>>> i2c.scan()
[]
>>>
```

I am running MicroPython v1.10-150-gf8f272429 on 2019-03-03; PCA10056 with NRF52840.

Thanks!

glennrub, 2019-03-13:
Hi @noblerabbit,

I did a dry-run of the I2C to see if there was something, but for me it is working. Could you try to define hardware instance 0 as first parameter in the I2C class constructor?

My output:
```
>>> MicroPython v1.10-213-g28c2873d9 on 2019-03-13; PCA10040 with NRF52832
Type "help()" for more information.
>>> import machine 
>>> i = machine.I2C(0, scl=machine.Pin(4), sda=machine.Pin(3))
>>> i.scan()
[60]
>>> 
```

noblerabbit, 2019-03-14:
Hi @glennrub,
Thanks so much for your reply! When passing in 0 as the first parameter to the I2C method as you suggested, the sensor now works. I can use pretty much the same driver as I am using with esp32 (the minor difference is that my nrf port does not have time.sleep() method, nor the time.ticks_us()...).

Closing the issue. Thanks again.

closed 2019-03-14 by noblerabbit