micropython/micropython #4575: rapid writes corrupts SDCard

rhubarbdog, 2019-03-05:
I've started with a newly formatted SD Card, running on a pyboard 1.1 with a SKIPSD file on the flash.
i've written a data logger which logs data in a callback on timer 4 running at 8kHz and writes it to SD Card.
When I plug the pyboard in the program runs to completion I think, I can't start a serial terminal fast enough to catch any errors.
but
```
>>> os.listdir()
['data-set.bin']
>>> os.rename('data-set.bin', 'data-set.rst')
```
I now soft reboot it to run again
```
PYB: sync filesystems
PYB: soft reboot
uncaught exception in Timer(4) interrupt handler
Traceback (most recent call last):
  File "main.py", line 36, in buffer_data
OSError: [Errno 22] EINVAL
MicroPython v1.9.4-543-gdc77fdb7d on 2018-09-19; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> os.listdir()
['data-set.bin', 'data-set.rst']
>>> data_set.terminate()
>>> os.umount()
```
ironically it looks like its worked, so I unplug the SD card and plug it into my laptop the disk looks blank.
so i put the SD Card back into my pyboard and press reset the program runs

I now start REPL and type `os.listdir()`
i now get 
```
['data-set.bin', 'data-set.rst', '\x08\x05\x00\x00\x1fT\x0f', '+\x05\x00\x1cv\x0f', ')\x05\x00\x00VX\x0f', '0\x03\x00\x00\x1bo\x0f', 'g\x03\x00\x15Q\x0f']
```
I unmount the SD card and remove it, it still looks blank in my laptop

here's a copy of the program i'm running

[data_logger2.py.txt](https://github.com/micropython/micropython/files/2930840/data_logger2.py.txt)

robert-hh, 2019-03-05:
Two aspects: 
a) About the PC reading the SD: could it read the SD before you ran your test?
b) Writing to a file in a callback function looks not good to me. The writing may take longer than the time between two calls. I would prefer to use a ring buffer which is filled by the callback and the writing is at "main" level, getting data chunks from the ring buffer and writing it to the file. You must in addition use some kind of lock mechanism to avoid concurrent access to the ring buffer.

rhubarbdog, 2019-03-05:
The SD card was fine before i started with this data logger. I lost some data (a whole directory). Before this attempt i had a partially complete file `data-set.bin` some binary named files. The file name `data-set.bin` had been corrupted, but i pressed f2 and renamed it.  I couldn't see the binary names on my laptop and couldn't delete them from the SD card on my pyboard. So i reformatted the disk to recover lost space.

dhylands, 2019-03-05:
Did you disable usb Mass storage on the pyboard? If not then that could also be corrupting things.

rhubarbdog, 2019-03-05:
I reformatted my SD card
edited boot.py so my pyboard is a serial device and mouse
copied main.py to my SD Card
I plug the pyboard in for the first time. It seems to go ok. I start REPL, when i type `os.listdir('/sd')` it returns `['main.py', 'data-set.bin']`.  I rename `data-set.bin` to `data.001` and soft reboot
I get the following
```
PYB: sync filesystems
PYB: soft reboot
uncaught exception in Timer(4) interrupt handler
Traceback (most recent call last):
  File "main.py", line 36, in buffer_data
OSError: [Errno 22] EINVAL
MicroPython v1.9.4-543-gdc77fdb7d on 2018-09-19; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> os.listdir()
['main.py', 'data-set.bin']
>>> os.rename('data-set.bin', 'data.001')
>>> 
PYB: sync filesystems
PYB: soft reboot
uncaught exception in Timer(4) interrupt handler
Traceback (most recent call last):
  File "main.py", line 36, in buffer_data
OSError: [Errno 22] EINVAL
MicroPython v1.9.4-543-gdc77fdb7d on 2018-09-19; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> os.listdir()
['main.py', 'data-set.bin', '5\x05\x00\x17M\x0f', '7\x05\x00\x00TO\x0f']
```
File `data.001` has been lost and 2 garbage file names are created

if what @robert-hh says regarding the callback taking longer to run than 2 succesive calls to it. surely not the desired result is something like exception `OSError : EBUSY` and no corruption to the file system

rhubarbdog, 2019-03-06:
I'm going to close this and open a new issue with the actual bug.
`OSError : EINVAL` is correct as the file has closed, but the callback hasn't been terminated.

closed 2019-03-06 by rhubarbdog