micropython/micropython #4576: unix port missing utime.mktime()

klapauciusisgreat, 2019-03-06:
It looks like the unix port has utime.mktime() missing

I was trying to make a script running both on micropython/regular python via the usual hack:

```
try: 
  import utime as time
except:
  import time
```
but since mktime is not in utime (as the documentation suggests), that fails.

klapauciusisgreat, 2019-03-06:
Also, I noted that utime does not have the dst flag in the time_tuple, which means it's incompatible with time as far as mktime is concerned (and others).

Would it be better to add dst to the tuple but ignore it ? I don't know how to gauge the trade-off here, I guess that it's probably not a common use case to run the same script in the various pythons and hence not worth the effort to 'fix'.

pfalcon, 2019-03-06:
It's not "missing", it's not supposed to be there. As you rightfully noticed, MicroPython's `utime` is not the same as CPython's `time`. If you need to run the same script on different implementations, you'd need to use either `time` or `utime` on both. For MicroPython, implementation of various modules beyond the builtin are provided by https://github.com/pfalcon/micropython-lib .

klapauciusisgreat, 2019-03-06:
I'm not sure I follow. Why is everything else in the unix port utime implementation except mktime ? (which is, btw, present in the esp866 and other ports, and also correctly dealing with dst wart I mentioned)

At the very least, the documentation should mention with parts of utime are not available on the unix port.

To me, it just looks someone forgot to implement the mktime call.

pfalcon, 2019-03-06:
> Why is everything else in the unix port utime implementation except mktime ?

"Everything else" - what? A lot of things are not there comparing with CPython's https://docs.python.org/3/library/time.html module. And the purpose of `utime` to be as minimal as possible, not any other way around. For example, localtime() was added to allow, after all, to print human-readable timestamps even without additional modules. And that's where it stops. Any extra functionality can be provided by external modules.

> (which is, btw, present in the esp866 and other ports, and also correctly dealing with dst wart I mentioned)

Unfortunately, some ports go beyond the bare minimum with the idea of giving *users of specific ports* additional conveniences. Unfortunately, because it leads to the opposite effect of some users thinking there's something "missing" in ports which don't provide them. Regarding esp8266 and other ports, it's hard to believe that it's implemented "correctly", because the correct implementation requires always-powered RTC, Epoch starting 1970-01-01 00:00 UTC, and double-precision floating point support.

> At the very least, the documentation should mention with parts of utime are not available on the unix port.

Not parts of the baseline API and thus not available on some ports, yeah. The unix port is just the guardian of the baseline API.

klapauciusisgreat, 2019-03-06:
I'm not arguing for implementing anything that isn't in the documentation [1]. I could even understand not implementing mktime in utime and removing it from the documentation. 
Maybe we should ask ourselves what is the documentation [1]  documenting, then? 

a) The baseline API? Then mktime should be implemented in all ports.
b) An arbitrary platform, e.g. the pyboard ? Then I argue the documentation should make that clear and ALSO specify that the baseline API actually is.

BTW, I realize that it's easy for to come in and act high and mighty without lifting a finger to help. That's not my intention at all.

[1] https://docs.micropython.org/en/latest/library/utime.html

pfalcon, 2019-03-07:
> I'm not arguing for implementing anything that isn't in the documentation [1].

You rightfully point at the issue in docs, thanks. I submitted a patch: https://github.com/micropython/micropython/pull/4583.
 
> I could even understand not implementing mktime in utime and removing it from the documentation.

Yeah, that definitely should be offered as an alternative to everyone wanting to  add more and more (uncontrollable) features, and using "another port has it" as an execute.

> Maybe we should ask ourselves what is the documentation [1] documenting, then?

During 5 years of project history that (and many other things) certainly was asked, discussed, and there're even decisions. Condensed summary of that is what I present here. (I'd encourage to use git history and ticket search for more.)

> a) The baseline API? Then mktime should be implemented in all ports.
b) An arbitrary platform, e.g. the pyboard ? Then I argue the documentation should make that clear and ALSO specify that the baseline API actually is.

The days when docs described arbitrary platform are long over. They now describe "generic MicroPython API" - both the baseline and common, platform-independent extensions. Any board-specific docs live in separate chapters like http://docs.micropython.org/en/latest/esp8266/quickref.html . But yeah, docs still has remnants of board-specific things and not all of the baseline API is marked as such. But intro at http://docs.micropython.org/en/latest/library/index.html should give the basic idea and set the expectations right.

dpgeorge, 2019-12-28:
The unix port has `uos.mktime()` as of 1b844e908c3a867be1e54efdb86a04c6a62b2f26

closed 2019-12-28 by dpgeorge