micropython/micropython #4577: [ESP32] About reading sensor value of ICM-20948 using ESP32-WROVER-B's I2C

KKawase0104, 2019-03-06:
Hello

I have coded a MicroPython script to read sensor values of ICM-20948 using I2C on ESP32-WROVER-B. 
Sometimes the value of the sensor was not correctly, but by changing exmod/machine_i2c.c the following it have read correctly.

>self->us_delay = 1000000 / freq;

As exmode/machine_i2c.c is also used other port, Is there a way to not change exmod/machine_i2c.c ?

Thanks

petrkr, 2019-03-06:
ESP32 I2C is software based. You can try lower frequency in constructor of I2C object, default seems to be 400kHz, Our I2C Expander pcf supports only 100kHz. I had to lower it too. But documentation of ICM says it should support 400kHz through.

`i2c = I2C(scl=i2c_scl, sda=i2c_sda, freq=100000) # 100kHz as Expander is slow :(`

KKawase0104, 2019-03-13:
Thank you for your fast reply. I'm sorry my late reply...
I tried to your advice and I could get correct sensor value.
Thank you.

I think #4595 relate on my problem.
I set the freq argument of the I2C constructor to 400000, 
so the SCL clock became 750 KHz. The ICM-20948 could 
not get the correct sensor value because it is up to 400kHz.

closed 2019-03-13 by KKawase0104