micropython/micropython #4578: Writing to a file in a callback corrupts the SD card when the file_id_object has closed

rhubarbdog, 2019-03-06:
When running the following program which writes to SD card in a timer's callback.  The SD card may become corrupt if the file has been closed before the callback has been set to `None`.
I discoverd this in a real example, but have written this version which you press the USR switch to close the file.

[corrupt.py.txt](https://github.com/micropython/micropython/files/2934078/corrupt.py.txt)

I'm running a pyboard version 1.1 and have edited my `boot.py` so the pyboard is a serial device and mouse.
Using a newly formatted SD Card, I insert the card into the pyboard and plug the pyboard into my laptop. I let the program run to completion.
I then start a REPL session and do a  soft reboot. I press the USR switch to close the file during execution. The first time i do this everything looks ok. So I repeat the soft reboot and press the switch again, this time all my data has been lost there's only `main.py` left on the SD card.
```
PYB: sync filesystems
PYB: soft reboot
uncaught exception in Timer(4) interrupt handler
Traceback (most recent call last):
  File "main.py", line 28, in timer_cb
OSError: [Errno 22] EINVAL
MicroPython v1.9.4-543-gdc77fdb7d on 2018-09-19; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> os.listdir()
['main.py', 'data-set.bin', 'data.001', 'new-file.bin']
>>> os.rename('new-file.bin', 'new.001')
>>> 
PYB: sync filesystems
PYB: soft reboot
uncaught exception in Timer(4) interrupt handler
Traceback (most recent call last):
  File "main.py", line 28, in timer_cb
OSError: [Errno 22] EINVAL
MicroPython v1.9.4-543-gdc77fdb7d on 2018-09-19; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> os.listdir()
['main.py']
```

rhubarbdog, 2019-03-06:
I first experienced this with the code from <a href="https://github.com/micropython/micropython/issues/4575">#4575</a> `data-logger2.py`.  When the bug is exhibited in a class it seems to cause more damage to the SD card

pfalcon, 2019-03-06:
You can't do anything in any interrupt callback beyond storing some data in pre-allocated data structures (e.g. an array) or update hardware registers.

dpgeorge, 2019-03-07:
It's not a problem to write to a file in a Timer or ExtInt (switch callback) IRQ handler, because the file IO IRQs are at a higher priority.  The only issue here is that the Timer callback is preempting the switch callback, and so it's trying to write to the file during the file close operation.

So the example script given above just needs some simple locking mechanism to get it working, like:
```python
def switch_cb():
    global file_id
    global open_

    open_ = False # set flag as closed before actually closing
    file_id.close()

def timer_cb(_):
    global file_id
    global buffer_
    global open_

    if open_:
        file_id.write(buffer_)
```
Note that 8000Hz may be too fast for an SD card (it was for the card I tested) so the timer_cb is running almost continuously.

An alternative to being careful with locks is to use `micropython.schedule(f, arg)` in the callbacks to schedule the actual file IO to happen at the lowest priority (where you can allocate memory, for example).

rhubarbdog, 2019-03-07:
Do you see the SD card corruption as a bug. I've seen loss of files and directories, capitaliseation of file names, generation of 'good' filenames and file names full of hexadecimal escapes.  The program is correctly creating exception `OSError : EINVAL` , but some or many writes are being made to a closing or closed file handle damaging the SD card.

dpgeorge, 2019-03-07:
> Do you see the SD card corruption as a bug.

Not really.  If you are doing file IO in a (hard) IRQ handler then you need to be careful with mutual access of the same resource.  

> The program is correctly creating exception `OSError : EINVAL`

It's probably not doing the right thing here, because it (file write) has definitely preempted a file close operation.  The error should be something like "file object already in use".

It's possible to make this work more cleanly, by enabling the locking feature of the fatfs library.  But it would only help to detect errors more easily, not enable to do shared atomic access to a file.  That's because the system is interrupt based, not thread based.