micropython/micropython #4581: [ESP32] How to use all 13 bits (8192 levels) of a PWM duty cycle?

delivrance, 2019-03-06:
According to [espressif docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/ledc.html#supported-range-of-frequency-and-duty-resolution), for PWM frequencies up to 5 kHz we can have `8192 = 2 ** 13` levels for PWM duty settings, but in micropython 1.10 it seems to be capped to `1024 = 2 ** 10` levels.

petrkr, 2019-03-06:
According to PWM code of uPy, it's compatibility with ESP8266

https://github.com/micropython/micropython/blob/master/ports/esp32/machine_pwm.c#L54

Mybe there could be some optional constructor which can by default use 10bits and by option change it to other.

dpgeorge, 2019-05-09:
> According to PWM code of uPy, it's compatibility with ESP8266

Yes, the `pwm.duty()` method is made to take the same 0-1023 range on both esp32 and esp8266.

It would be good to improve it to take advantage of more resolution.  See #4237 for a proposal for an improved PWM API that can accommodate this.

IhorNehrutsa, 2022-01-14:
[PWM.duty_u16](https://docs.micropython.org/en/latest/library/machine.PWM.html#machine.PWM.duty_u16) solves this issue

@dpgeorge please close

closed 2022-01-17 by dpgeorge