micropython/micropython #4586: [mboot] Unable to download firmware using mboot

Meekdai, 2019-03-07:
I successfully deploy mboot on PYBV10 according to the usage of mboot/README.md. The red LED is blink once per second. In the Dfuse software, I can see that the chip has entered the dfu mode.
But when I download the micropython firmware, the chip can't be erased.

```
1 Device(s) found :
Device [1]: STM Device in DFU Mode, having [1] alternate targets
Target 00: Upgrading - Erase Phase (0)... Duration: 00:00:49
```

dpgeorge, 2019-03-07:
> But when I download the micropython firmware, the chip can't be erased.

You need to also build the MicroPython firmware with mboot enabled, like `make USE_MBOOT=1`.  This ensures that the MicroPython firmware starts at address 0x08020000 instead of the usual 0x08000000 (this is explained in mboot/README.md).

Meekdai, 2019-03-07:
I use address `TEXT0_ADDR = 0x08004000` and command `make BOARD = PYBV10 USE_MBOOT = 1`. I am using Windows, I suspect that the problem is caused by Dfuse software.

dpgeorge, 2019-03-07:
mboot doesn't (yet) support mass erase, and will not allow you to erase/program the sector that it lives in (starting at 0x08000000).  The dfu-util command-link program is known to work with mboot, and also tools/pydfu.py in this repository.  It could be that Dfuse is sending some commands over USB that mboot does not support/recognise.

Meekdai, 2019-03-11:
I successfully flash the micropython firmware by using dfu-util, but it doesn't seem to work.Is there a problem with my configuration?
mpconfigboard.mk
```
MCU_SERIES = f4
CMSIS_MCU = STM32F405xx
AF_FILE = boards/stm32f405_af.csv
LD_FILES = boards/stm32f405.ld boards/common_ifs.ld
TEXT0_ADDR = 0x08004000
TEXT1_ADDR = 0x08020000
```
Does stm32f405.ld need to be modified?

dpgeorge, 2019-03-11:
>  but it doesn't seem to work.Is there a problem with my configuration?

Yes there are problems with the config: the filesystem goes at 0x08004000 so you can't put TEXT0 there.  Also, common_ifs.ld won't work, it needs common_blifs.ld instead (bootloader + internal filesystem config).

Please see ports/stm32/boards/PYBV10/mpconfigboard.mk, it has a working configuration when USE_MBOOT=1.

Meekdai, 2019-03-12:
Thanks damien, it works on my PYBV10 board. 
I saw that mboot supports SPI FLASH. I want to download the image file to the external flash memory via the network and update it via mboot. My pyboard use the W5500 and works well. Are there any examples for reference?

dpgeorge, 2019-03-12:
>  I want to download the image file to the external flash memory via the network and update it via mboot. My pyboard use the W5500 and works well. Are there any examples for reference?

Yes, see `ports/stm32/mboot/fwupdate.py`, in particular the `update_mpy()` function.  All you need to do is have a firmware file in the filesystem in .dfu.gz form and mboot will be able to program it automatically (note: `fs_base` and `fs_len` specify the memory-mapped location of the FAT filesystem for mboot to find it).

Meekdai, 2019-03-19:
I added SPI FLASH (W25Q64) to my PYBV10 and successfully expanded the flash. Refer to the configuration of the STM32L476DISC.
```
// use external SPI flash for storage
#define MICROPY_HW_ENABLE_INTERNAL_FLASH_STORAGE (0)
#define MICROPY_HW_SPIFLASH_SIZE_BITS (64 * 1024 * 1024)
#define MICROPY_HW_SPIFLASH_CS      (pin_A2)
#define MICROPY_HW_SPIFLASH_SCK     (pin_A5)
#define MICROPY_HW_SPIFLASH_MOSI    (pin_A7)
#define MICROPY_HW_SPIFLASH_MISO    (pin_A6)

// block device config for SPI flash
extern const struct _mp_spiflash_config_t spiflash_config;
extern struct _spi_bdev_t spi_bdev;
#define MICROPY_HW_BDEV_IOCTL(op, arg) ( \
    (op) == BDEV_IOCTL_NUM_BLOCKS ? (MICROPY_HW_SPIFLASH_SIZE_BITS / 8 / FLASH_BLOCK_SIZE) : \
    (op) == BDEV_IOCTL_INIT ? spi_bdev_ioctl(&spi_bdev, (op), (uint32_t)&spiflash_config) : \
    spi_bdev_ioctl(&spi_bdev, (op), (arg)) \
)
#define MICROPY_HW_BDEV_READBLOCKS(dest, bl, n) spi_bdev_readblocks(&spi_bdev, (dest), (bl), (n))
#define MICROPY_HW_BDEV_WRITEBLOCKS(src, bl, n) spi_bdev_writeblocks(&spi_bdev, (src), (bl), (n))

```
I added the definition `#define MBOOT_FSLOAD (1)`, and recompile the firmware of mboot and micropython. But I don't know how to passing data elements. Is it possible to write a function that can be called directly to enter mboot and automatically update the firmware?

Meekdai, 2019-03-19:
I added SPI FLASH (W25Q64) to my PYBV10 and successfully expanded the flash refer to the configuration of the STM32L476DISC.
```
// use external SPI flash for storage
#define MICROPY_HW_ENABLE_INTERNAL_FLASH_STORAGE (0)
#define MICROPY_HW_SPIFLASH_SIZE_BITS (64 * 1024 * 1024)
#define MICROPY_HW_SPIFLASH_CS      (pin_A2)
#define MICROPY_HW_SPIFLASH_SCK     (pin_A5)
#define MICROPY_HW_SPIFLASH_MOSI    (pin_A7)
#define MICROPY_HW_SPIFLASH_MISO    (pin_A6)

// block device config for SPI flash
extern const struct _mp_spiflash_config_t spiflash_config;
extern struct _spi_bdev_t spi_bdev;
#define MICROPY_HW_BDEV_IOCTL(op, arg) ( \
    (op) == BDEV_IOCTL_NUM_BLOCKS ? (MICROPY_HW_SPIFLASH_SIZE_BITS / 8 / FLASH_BLOCK_SIZE) : \
    (op) == BDEV_IOCTL_INIT ? spi_bdev_ioctl(&spi_bdev, (op), (uint32_t)&spiflash_config) : \
    spi_bdev_ioctl(&spi_bdev, (op), (arg)) \
)
#define MICROPY_HW_BDEV_READBLOCKS(dest, bl, n) spi_bdev_readblocks(&spi_bdev, (dest), (bl), (n))
#define MICROPY_HW_BDEV_WRITEBLOCKS(src, bl, n) spi_bdev_writeblocks(&spi_bdev, (src), (bl), (n))

```
I added the definition `#define MBOOT_FSLOAD (1)`, and recompile the firmware of mboot and micropython. Successfully deploy them and run mboot, use `dfu-util -l`
```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=2200, devnum=49, cfg=1, intf=0, path="1-6", alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg", serial="396E31693038"
```
Then go into micropython and run the code below.
```
import fwupdate
fwupdate.update_mpy('firmware.dfu.gz',0x08000000,0x1c000)
```
I have put file 'firmware.dfu.gz' in the root of the filesystem, but I am not sure if `fs_base` and `fs_len` are correct. The code is not working as expected.

Meekdai, 2019-03-20:
Successfully downloaded. I ignored the configuration of MBOOT in SPI FLASH.
```
// Bootloader configuration (only needed if Mboot is used)
#define MBOOT_FSLOAD (1)
#define MBOOT_SPIFLASH_ADDR (0x80000000) 
#define MBOOT_SPIFLASH_BYTE_SIZE (8 * 1024 * 1024) 
#define MBOOT_SPIFLASH_LAYOUT "/0x80000000/016*04Kg" 
#define MBOOT_SPIFLASH_ERASE_BLOCKS_PER_PAGE (16 / 4) 
#define MBOOT_SPIFLASH_SPIFLASH (&spi_bdev.spiflash) 
#define MBOOT_SPIFLASH_CONFIG (&spiflash_config)
```
Use the following code to update the firmware.

```
import fwupdate
fwupdate.update_mpy('firmware-mboot.dfu.gz',0x80000000,0x400000)
```

closed 2019-03-20 by Meekdai