micropython/micropython #4594: QR code generator

prusnak, 2019-03-10:
In our project, we have a QR code generator based on https://www.nayuki.io/page/qr-code-generator-library

Would there be an interest in having a native module for QR code encoding - `extmod/moduqr.c`? Optionally built of course.

I think a lot of people need this functionality and https://github.com/JASchilz/uQR is just too slow :-/

mattytrentini, 2019-03-14:
I'm interested in generating QR Codes in MicroPython. I had intended to try porting the [Python variant of Nayuki's QR code generator](https://github.com/nayuki/QR-Code-generator/tree/master/python) but I'm not yet sure if performance will be reasonable...

Have you had a chance to profile uQR? How slow is too slow? :)

So, yes, I'm keen to have a native module _if_ the performance of a pure MicroPython module is particularly poor. Happy to help out with the implementation.

peterhinch, 2019-03-14:
From a quick look at **uQR** there would seem to be scope for optimisation both for size and speed. For example
```python
G15 = (
    (1 << 10) | (1 << 8) | (1 << 5) | (1 << 4) | (1 << 2) | (1 << 1) |
    (1 << 0))
```
is an expensive way to assign 0x537 to a variable. There is at least one global dict which could be replaced by a lambda. The alternative code emitters could be used. The final output is an extravagant (in RAM) way to store a bitmap. And that's just from a quick scan of the code.

A couple of questions for anyone thinking of porting or optimising this:

-  Are there features which can be culled for a "micro" version?
- Should the output got into a **framebuf** instance?

All this boils down to "how slow is too slow". If it's >= an order of magnitude there's no choice than to write it in C (IMO). And before anyone suggests it, C programming is only something I do _in extremis_ so the answer is no. :)

prusnak, 2019-03-14:
@mattytrentini I haven't profiled Python version of Nayuki's QR code generator nor do I plan to do so. But the uQR from micropython-lib was indeed really slow, don't have the numbers though. Maybe if you do the benchmark of these two, we can decide whether it is worth having this code natively or not.

peterhinch, 2019-03-14:
I did a quick test of [JASchilz/uQR](https://github.com/JASchilz/uQR.git) on a Pyboard 1.0. Compilation failed with a memory error, so I ran with frozen bytecode. I would expect that a precompiled mpy would have worked too.
```python
from uQR import QRCode
import utime
import micropython, gc
gc.collect()
print(micropython.mem_info())
t = utime.ticks_ms()
qr = QRCode()
dt1 = utime.ticks_diff(utime.ticks_ms(), t)
t = utime.ticks_ms()
qr.add_data('uQR rocks!')
dt2 = utime.ticks_diff(utime.ticks_ms(), t)
t = utime.ticks_ms()
matrix = qr.get_matrix()
dt3 = utime.ticks_diff(utime.ticks_ms(), t)
print(dt1, dt2, dt3)
gc.collect()
print(micropython.mem_info())
```
The outcome was:
```
dt1 == 1ms
dt2 == 0ms
dt3 == 464ms
RAM use 6.3KiB
```
I then repeated the test, changing the string to ```'uQR rocks! This is a longer string more like a typical URL'``` with the following outcome:
```
dt1 == 0ms
dt2 == 2ms
dt3 == 1200ms
RAM use 13KiB
```
RAM use is getting excessive, but this is largely due to the inefficient format of ```matrix``` which I estimate to be using ```41*41*4 = 6.7KiB``` to store ```41*6 = 246 bytes``` of information (6 bytes is more than sufficient to store 41 bits).

The code looks like a port of "big iron" Python to MicroPython with no attempt at optimisation. How much of an improvement can be made to speed would need proper profiling. But I have no idea what speed you guys need. If you need more than 10x my "finger in the air" guess is that you're best off going to C. Less and we might be in with a chance with optimisation and I may be able to help.

What are we expecting here?

prusnak, 2019-03-15:
It would be interesting to see the same kind of benchmark @peterhinch did but with the Nayuki's code. Anyone?

peterhinch, 2019-03-15:
@prusnak 
If you mean [this source file](https://github.com/nayuki/QR-Code-generator/blob/master/python/qrcodegen.py) it would need porting to MicroPython first.

prusnak, 2019-03-15:
@peterhinch yes

peterhinch, 2019-03-16:
Alas I'm concluding that this project is dead in the water. I'm losing interest.

Nobody has responded to my requests for a performance target or options for a 'micro' feature subset; still less expressed any interest in actually contributing effort.

I can port, profile and optimise code. Without agreed objectives and help from people with an understanding of the code design my efforts would probably be wasted. I don't plan to tackle this as a solo project.

mattytrentini, 2019-03-18:
Apologies @peterhinch, I've been crazy-busy at work and was also away this weekend. Even so, I have a long list of MicroPython work I want to tackle and QR codes are unfortunately not the highest priority at the moment...I really appreciate you looking into this though!

My performance needs are modest; halving those times would be quite usable for me (0.5-1 sec for reasonably-sized QR codes would be manageable - though my gut says that's _slow_). I agree that there appears to be some significant improvements possible. I'd be most likely using the ESP32 or possibly an STM32F4. _Not_ an ESP8266. Framebuf would be a logical render target but I'm also curious about how it could be integrated into LittlevGL as that shows promise as a GUI toolkit.

But I need to be careful about over-committing; I really can't commit to doing much with QR codes right now...perhaps @prusnak might have more time!

peterhinch, 2019-03-18:
One potential optimisation is to use Native or Viper code for performance critical functions. This is [unavailable on ESP32](https://github.com/micropython/micropython/issues/4607#issuecomment-473743801). My gut feeling (without having done any profiling) is that a 2:1 improvement will be readily available even without recourse to native code. On a Pyboard D with appropriate use of native code it might be considerably quicker...

petrkr, 2019-04-02:
Did only call tests, not profiling code.

```
get_matrix took: 1884ms
set_frame_memory took: 84ms
display_frame took: 2ms
efb_show took: 92ms
display_qr took: 371ms
```

but already did already few optimizations like 
```
-        return lambda i, j: (int(i / 2) + int(j / 3)) % 2 == 0
+        return lambda i, j: (i // 2 + j // 3) % 2 == 0
```
which give 1.8sec from 1.9sec

mattytrentini, 2019-07-15:
I revisited this topic and ported the Nayuki Project's Python code to MicroPython. Also [documented the porting effort](https://github.com/mattytrentini/QR-Code-generator/tree/master/python) in the hope it may be useful for others porting from Python to MicroPython. 

I've not made _any_ performance improvements, simply ported the code so the identical codebase works on both Python and MicroPython. 

And performance, on my [TinyPICO](https://www.crowdsupply.com/unexpected-maker/tinypico) is _slow_: typically ~1.9 seconds to generate a QR code with a short string! 

There are quite a few obvious performance improvements to be made. That said, I'm giving a talk at PyCon AU: [Extending MicroPython: Using C for good!](https://2019.pycon-au.org/talks/extending-micropython-using-c-for-good) in a few weeks so my next steps are to build the C port into a MicroPython module,  explain how that's done and then measure it...

I'll have to revisit performance improvements after PyCon AU! If someone else wants to see how fast they can make Nayuki then by all means give it a go; it's at least running on MicroPython now... :)

nayuki, 2019-07-21:
@mattytrentini: Thanks for your MicroPython porting work! I apologize for the trouble you went through because my Python library is not optimized for this use case.

To be clear, my Python QR Code lib is designed to prioritize absolute correctness, feature-completeness, human-reading clarity and conciseness, and then performance last. As one example, [`reed_solomon_multiply()`](https://github.com/nayuki/QR-Code-generator/blob/1e24fcf67a0d876f3d6e1b1a88a8313e3beee62f/python/qrcodegen.py#L602-L614) is implemented as a slow but understandable algorithm instead of as an opaque 256×256 lookup table.

Python, as an interpreted language, is already demonstrably slow on desktop computers. When doing pure numeric and array operations (as opposed to I/O or call C modules), Python is roughly [30× slower](https://www.nayuki.io/page/project-euler-solutions#benchmark-timings) than a compiled language like Java which has fixed-size integers.

For the microcontrollers I worked with, they are about 100× slower than desktop computers. This exacerbates Python's slowness; also their small memory may not be appropriate for running a Python interpreter. So specifically for embedded systems, I advocate using the [C version](https://github.com/nayuki/QR-Code-generator/tree/master/c) of my QR Code generator library ([photo](https://www.nayuki.io/res/qr-code-generator-library/teensy-qrcodegen.jpg), [video](https://www.nayuki.io/res/qr-code-generator-library/teensy-qrcodegen.mp4)), which is very streamlined in terms of CPU time and memory usage. On a side note, the Moddable project [packaged up](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/data/qrcode/qrcodegen.c) my C lib into their JavaScript framework.

peter-conalgo, 2020-06-08:
I've taken @nayuki 's C library and wrapped it into a module. I tried to keep the API simple and still powerful enough for the things I want to do, like control the version number precisely.

Needs just a little more to make it a dynamically-loadable native-code `.mpy`, but nearly there. Works great compiled into Micropython, both versions 1.12 and 1.9.4 (tested).

[mpy-qr](https://github.com/coinkite/mpy-qr)

TerryIoannou, 2023-12-22:
Would it be better just create an API on your server that just generated a qrcode and then send it to your Microboard as a png and  then displays it within the dimention of the screen you have connected with your microboard ?

and from the microboard send data back to the api after its scanned or am I missing something ?

petrkr, 2023-12-22:
> Would it be better just create an API on your server that just generated a qrcode and then send it to your Microboard as a png and then displays it within the dimention of the screen you have [...]

Very good idea, but tell me, how exactly this will work on offline device?

TerryIoannou, 2023-12-23:
I have not thought of that actually which makes me wonder more about the current project I am working on thank you for the thought xD if you have the solution to it then that would be perfect!

peterhinch, 2023-12-23:
See [the QRMap widget](https://github.com/peterhinch/micropython-micro-gui/blob/main/README.md#620-qrmap-widget) in [MicroGui](https://github.com/peterhinch/micropython-micro-gui/tree/main).

jonnor, 2024-07-02:
There seems to be several good QR code alternatives here. Perhaps this issue can be closed now? Maybe use a discussion if further discussion is wanted.

Just triaging a bit to maybe one day get below 1k open issues :D

petrkr, 2024-07-03:
> There seems to be several good QR code alternatives here. Perhaps this issue can be closed now? Maybe use a discussion if further discussion is wanted.
> 
> Just triaging a bit to maybe one day get below 1k open issues :D

That alternatives are very slow.
Tooks like 2 sec to generate and display QR code.

Prusnak posted C codec which do everything in just 300ms.

So perhaps you can close it whenever there will be solution how to use those optimized libraries.

There us QR code, which is compiled as C user module, but that is part of BIN, would be nice if it can be compiled as 'so' library and put on filesystem like mpy can

petrkr, 2024-07-03:
Talking about this one 

https://github.com/coinkite/mpy-qr

mattytrentini, 2024-07-03:
I gave a (admittedly not my finest!) talk a few years ago about using a native module to generate QR codes: [Extending MicroPython: Using C for good!](https://www.youtube.com/watch?v=437CZBnK8vI). There is a [gist](https://gist.github.com/mattytrentini/09885f5aae78590e5d4cce15a18c62dc) that contains the guts of the example. Generating QR codes took the same as when using the underlying C library; 11ms.

Perhaps I should polish that up and make a proper library out of it...