micropython/micropython #4595: ESP32 I2C speed issue

douggilliland, 2019-03-11:
Setting the I2C speed on the I2C for the ESP32 results in seriously different clock rates than the parameter.

Setting freq=100000 is close.
Setting freq=200000 is 328 KHz
Setting freq-400000 is 750 KHz.

freq=400000 waveform capturted at:
http://land-boards.com/blwiki/index.php?title=ESP32#I2C_Speed

dpgeorge, 2019-03-11:
Thanks for the report.  I guess that `ports/esp32/mphalport.c:mp_hal_delay_us_fast()` needs to be improved because that is responsible for the timing (it's a software based I2C implementation).

douggilliland, 2019-03-11:
It was a fun one to find since I have a card with 8 MCP23017 parts which apparently runs just fine at 750KHz but my card with just two MCP23017 parts doesn't.

Thanks for the quick response. Is there any way to use the Hardware I2C instead? The ESP32 does support hardware I2C.

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/i2c.html

By the way, I love MicroPython. Thank you!

dpgeorge, 2019-03-12:
In 297092a76a6be2d5f5da32524e91f768e95937f4 I switched `mp_hal_delay_us_fast()` to use the system provided `ets_delay_us()` which is more accurate.  So now the I2C should not go too fast (but note that the granularity of frequencies is not great, the formula is `delay_half_us=500000/freq` so the valid high frequencies are 500000, 250000, 166666, 125000, 100000, etc).

> Is there any way to use the Hardware I2C instead?

Not yet, it needs implementing.

> By the way, I love MicroPython. Thank you!

Great to hear!

KKawase0104, 2019-03-14:
I'm MicroPython source code beginner, so if I write wrong question, sorry.

I wonder why as soft I2C freq sets 400000 ,SCL become 750kHz?
Is it the specification ?

Thank you.

peterhinch, 2019-03-14:
If you read each message in this thread you will see the almost certain reason. The problem was fixed two days ago so you'll need to update your firmware to use the latest daily build.

douggilliland, 2019-03-14:
@KKawase0104  He had a software loop in there which was processor speed dependent.

KKawase0104, 2019-03-15:
@peterhinch @douggilliland 
I appreciate your reply.
I retried using updated firmware and got correct sensor value.
I'm sorry I took your time for my carelessness.
Thank you.

kevinkk525, 2019-03-27:
As far as I know loboris got the i2c hardware interface integrated, so if someone wants to put some effort in it, it should be easy to backport (I hope)

dpgeorge, 2019-07-10:
See #4914 for hardware I2C on the esp32.

dpgeorge, 2019-07-19:
Hardware I2C for esp32 was implemented it a29334761dc7b8eaf9f78611e3cd9033928b110c

closed 2019-07-19 by dpgeorge