micropython/micropython #4596: [STM32][NUCLEO_401RE] Dropping characters in REPL while typing

rolandvs, 2019-03-12:
Just been testing with the NUCLEO_401RE.
When (trying) entering commands in REPL characters get lost.

It is a basic board with the UART going through the STLINK VCOM.

Bringing down the speed of the REPL port doesn't help. It looks like it is something in the input buffer handling.

dpgeorge, 2019-03-13:
It could have been introduced in f938e70c6905474b08f7b5e6df3da9c1b0dcf01a

Can you revert to e61862d063b7a19f5baba8e0d1234d648826a616 (prior commit to the above) and test if the issue is still there?

rolandvs, 2019-03-13:
With HSI enabled and a _real power cycle_ reset: it was working as expected.

So did the problem go? No, after issuing sometimes one or two `machine.reset()` commands, the character dropping started again.

Alternatively I turned off the HSI with `#define MICROPY_HW_CLK_USE_HSI (0)` and everything is fine using the same test procedure as above.

chrismas9, 2019-03-13:
According to the 401 reference manual the HSICAL bits in the RCC_CR register are undefined after reset. They are "initialised automatically at startup".

Startup is not defined, but I get the impression it is related to a Vdd event (power on) or wakeup from stop or sleep. If this is the case power on would calibrate the HSI and software reset could change the HSICAL bits to anything (undefined). This would explain the observed behaviour. It might be worth reloading the HSICAL bits after a software reset and maybe after wakeup to be safe.

dpgeorge, 2019-03-13:
Ok, it sounds like it's definitely an issue with HSI calibration leading to a bad UART baudrate.

rolandvs, 2019-03-23:
Need to test it but in `system_stm32.c` I added ` RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;` for testing. Also I did not find any reference for `STM32F0`, which seems odd.... Need to check this too...

```
#if defined(STM32F4) || defined(STM32F7) || defined(STM32H7)
    RCC_OscInitStruct.OscillatorType = MICROPY_HW_RCC_OSCILLATOR_TYPE;
    RCC_OscInitStruct.HSEState = MICROPY_HW_RCC_HSE_STATE;
    RCC_OscInitStruct.HSIState = MICROPY_HW_RCC_HSI_STATE;
    RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
```

dpgeorge, 2019-03-26:
Should be fixed by d396a7e10d69f9490ca3aaa0b2ca147223f20314

commit d396a7e1 (2019-03-26): stm32/system_stm32: Provide default value for HSI calibration.

If HSI is used the calibration value must be valid.  Fixes #4596.
closed 2019-03-26 by dpgeorge