micropython/micropython #4600: make BOARD=NUCLEO_F429ZI error

deeptuuk, 2019-03-13:
when i run the command make BOARD=NUCLEO_F429ZI 
i meet an error below:
make: *** No rule to make target 'lib/lwip/src/core/def.c', needed by 'build-NUCLEO_F429ZI/genhdr/qstr.i.last'.  Stop.

last:
how can i fix it;
please help me as soon as possible.
thank you very much

deeptuuk, 2019-03-13:
i have fix it
thank you very much

closed 2019-03-13 by deeptuuk

hybotics, 2022-11-12:
It would be helpful if you would outline how you fixed this for the benefit of those who come after you. :)

jimmo, 2022-11-14:
> @deeptuuk It would be helpful if you would outline how you fixed this for the benefit of those who come after you. :)

Almost certainly the fix would have been `make BOARD=NUCLEO_F429ZI submodules`