micropython/micropython #4606: Unable to get mboot to work on PYBV11 (or PYBV10)

dhylands, 2019-03-14:
I'm just trying to reproduce using mboot on the PYBV11. I was using the latest master c9eb7eb449a246c607fa92a5313e3137d728fbcb 

I started by doing a regular micropython build:
```
cd ports/stm32
make BOARD=PYBV11 clean all deploy
```
which brought up Micropython fine.

I then put the board in DFU mode and built and flashed mboot
```
cd ports/stm32/mboot
make BOARD=PYBV11 clean all deploy
```
I used the USR switch to get mboot into mboot mode (red LED flashing) and dfu-util -l shows:
```
683 >dfu-util -l
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=2200, devnum=30, cfg=1, intf=0, alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg", serial="336F338F3433"
```
I then went back to the stm32 directory and build micropython for mboot:
```
cd ports/stm32
make BOARD=PYBV11 USE_MBOOT=1 clean all deploy
```
The flashing built a single segment image and it seemed to flash ok:
```
Writing build-PYBV11/firmware.dfu to the board
File: build-PYBV11/firmware.dfu
    b'DfuSe' v1, image size: 333373, targets: 1
    b'Target' 0, alt setting: 0, name: "ST...", size: 333088, elements: 1
      0, address: 0x08020000, size: 333080
    usb: 0483:df11, device: 0x0000, dfu: 0x011a, b'UFD', 16, 0x687e6871
Writing memory...
0x08020000  333080 [=========================] 100% 
Exiting DFU...
```
However, when I reset the board, I just get the 3 dimly lit LEDs. Pushing the USR button gets me back into DFU mode.

Before investigating further, I wanted to make sure that I'm not missing a step.

dpgeorge, 2019-03-15:
> I then put the board in DFU mode and built and flashed mboot

You need to build mboot also with the `USE_MBOOT=1` option to make.  This is because of the following line in mboot's Makefile:
```
CFLAGS += -DAPPLICATION_ADDR=$(TEXT0_ADDR)
```
That is, mboot must be built knowing where the main/application firmware starts.  And this is set in the board's mpconfigboard.mk file.  For PYBv1.x, since mboot is not the default, use must set `USE_MBOOT=1`.  For boards where mboot is the default (see eg STM32F769DISC) you don't need to explicitly set this.

Perhaps the mboot/REAME.md could do with some more explicit wording about this?

dhylands, 2019-03-15:
Perhaps the mboot Makefile should just set USE_MBOOT=1 in the Makefile? (and have a comment similar to what you mentioned above about why it's being set). It seems to me that this wouldn't cause any problems when it isn't really needed.

The README only mentions using USE_MBOOT=1 when building the microython firmware and the examples that are there don't use it show it (which is why I didn't realize it was needed).

I can confirm that everything works as expected when mboot is built with USE_MBOOT=1 is set.

dpgeorge, 2019-03-15:
> Perhaps the mboot Makefile should just set USE_MBOOT=1 in the Makefile?

Yes, that would work and be convenient.

> The README only mentions using USE_MBOOT=1 when building the microython firmware and the examples that are there don't show it

I'm sure I did add this, see point 1 in the example section.

dhylands, 2019-03-15:
> > The README only mentions using USE_MBOOT=1 when building the microython firmware and the examples that are there don't show it
> 
> I'm sure I did add this, see point 1 in the example section.

Yep - you're right. I'm not sure why I didn't see this. Anyways, I'll put together a PR to put USE_MBOOT=1 in the mboot Makefile.

closed 2019-03-15 by dhylands