micropython/micropython #4607: [ESP32] The Viper code emitter

kingsjl, 2019-03-15:
When does esp32 support the Viper code emitter

dpgeorge, 2019-03-18:
To add support for native code generation (native, viper, inline asm) to the esp32 port requires the following items:
- implement `nlr_push` as a function, not a macro, for the xtensa architecture with windowing
- add support to the native emitter for xtensa call/ret with windowed registers
- add a way to allocate, free and garbage collect executable RAM on the esp32 (RAM used for the uPy heap is not executable)

There is not currently a timeline to implement these features.

adritium, 2019-03-21:
Is this also true for ESP8266?

robert-hh, 2019-03-21:
ESP8266 supports native and viper (and assembly), but you have to take some precautions for memory usage. I have saved a few links:
https://github.com/micropython/micropython/issues/2672
https://github.com/micropython/micropython/commit/f76b1bfa9f59fcfa03837c6934ad51d2db3ff4a3

maludwig, 2019-08-26:
So I've just spent a long time trying to get this to work, until finally finding this thread.

I have a much easier request, can you make it so that instead of this output:
```
===
=== @micropython.native
=== def foo(arg):
===     return (arg * arg) + arg
===
=== print(foo(4))
===
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
SyntaxError: invalid micropython decorator
```

It instead says something like:
```
SyntaxError: native and viper compilers are not yet supported on the ESP32
```

It would make my life a lot better. Well...it would have.

dpgeorge, 2019-09-10:
See #5082 for native support en esp32.

maludwig, 2019-10-23:
Hey, so with the latest build on my ESP32, I'm getting the following oddity.
Viper works with raw ints.
If you index a bytes, you get an int `b'\x03'[0] == 3`
But if you return an indexed byte as an int with viper, it doesn't return, but instead explodes.

I'm betting that, like, somehow under the hood it's like, a `uint8_t`, and not an `int`. And this is confusing for viper. But I have no clue how to even begin to test that theory.

My use case is to basically recreate i2c.readfrom, but for the SPI class.

Output is below:

```
MicroPython v1.11-481-gf1d91908f on 2019-10-22; ESP32 module with ESP32
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
===
=== @micropython.viper
=== def f2(n: int) -> int:
===     return n*2
===
=== @micropython.native
=== def f3() -> int:
===     return (b'\x03'[0])*3
===
>>>
>>> type(b'\x04')
<class 'bytes'>
>>> type(b'\x04'[0])
<class 'int'>
>>>
>>> print(f2(2))
4
>>> print(f3())
9
>>>
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
===
=== @micropython.viper
=== def f4() -> int:
===     return (b'\x04'[0])*4
===
Traceback (most recent call last):
  File "<stdin>", line 3, in f4
ViperTypeError: return expected 'int' but got 'object'
>>>

```

robert-hh, 2019-10-23:
As far as I understand it, it is a name collision between Pyhton and viper about the word int. b'\x04'[0] is a python object of the class int. Wheres a Viper int object is just the raw bit pattern, like in C. The number 4 may be treated as both in viper code. In the function, you can cast b'\x04'[0] to Viper-int with int(b'\x04'[0]), and then it works.
In your example you multiply a Python object with a number, resulting in a Python object.

peterhinch, 2019-10-24:
> In your example you multiply a Python object with a number, resulting in a Python object.

I think when you multiply a Python object by a number you get an exception:
```python
import micropython
@micropython.viper
def f4() -> object:
    return (b'\x04'[0])*4

Traceback (most recent call last):
  File "<stdin>", line 3, in f4
ViperTypeError: can't do binary op between 'object' and 'int'
```
But your explanation is correct in that the following returns a Python `int` object (4):
```python
import micropython
@micropython.viper
def f4() -> object:
    return (b'\x04'[0])
```
Interesting.

amirgon, 2019-11-01:
Is there some limitation with viper/native emitters on frozen code?    

When I load my module from the filesystem it works fine, but when running it as a frozen module I get WDT timeout and "IllegalInstruction" errors....

peterhinch, 2019-11-02:
My guess is that the cross compiler isn't configured for the ESP32 instruction set when you perform a build. I'd look at the makefile.

dpgeorge, 2019-11-04:
> My guess is that the cross compiler isn't configured for the ESP32 instruction set when you perform a build.

Yes, good thought!  I fixed this in 742030945c779c57c30c62c7b1353758d29dc228, and also needed to fix mpy-tool.py in 36c9be6f60314684edf131f554efec7120e20f30 to get it to freeze correctly.

So now native/viper should be well supported on esp32, in .py, .mpy and frozen code.

amirgon, 2019-11-27:
I'm getting `error freezing mpy` when trying to freeze some module with Viper code.
The same module works fine when imported from the file system.

```
|| // frozen viper code for scope SPH0645__lt_module_gt__Recorder_
|| STATIC const byte fun_data_SPH0645__lt_module_gt__Recorder_[848] __attribute__((section(".text,\"ax\",@progbits # "))) = {
||     0x06, 0x05, 0x00, 0x00, 0x08, 0x00, 0x08, 0x00,
|| Traceback (most recent call last):
||   File "/home/amirgon/esp/projects/lv_mpy/tools/mpy-tool.py", line 834, in <module>
||     main()
||   File "/home/amirgon/esp/projects/lv_mpy/tools/mpy-tool.py", line 828, in main
||     freeze_mpy(base_qstrs, raw_codes)
||   File "/home/amirgon/esp/projects/lv_mpy/tools/mpy-tool.py", line 772, in freeze_mpy
||     rc.freeze(rc.source_file.str.replace('/', '_')[:-3] + '_')
||   File "/home/amirgon/esp/projects/lv_mpy/tools/mpy-tool.py", line 385, in freeze
||     self.freeze_children(parent_name)
||   File "/home/amirgon/esp/projects/lv_mpy/tools/mpy-tool.py", line 255, in freeze_children
||     rc.freeze(self.escaped_name + '_')
||   File "/home/amirgon/esp/projects/lv_mpy/tools/mpy-tool.py", line 385, in freeze
||     self.freeze_children(parent_name)
||   File "/home/amirgon/esp/projects/lv_mpy/tools/mpy-tool.py", line 255, in freeze_children
||     rc.freeze(self.escaped_name + '_')
||   File "/home/amirgon/esp/projects/lv_mpy/tools/mpy-tool.py", line 495, in freeze
||     i += self._link_qstr(i, qi_kind, qst)
||   File "/home/amirgon/esp/projects/lv_mpy/tools/mpy-tool.py", line 469, in _link_qstr
||     assert 0
|| AssertionError
|| 
py/mkrules.mk|103| recipe for target 'build-GENERIC_SPIRAM/frozen_content.c' failed
|| make: *** [build-GENERIC_SPIRAM/frozen_content.c] Error 1
|| make: Leaving directory '/home/amirgon/esp/projects/lv_mpy/ports/esp32'
```

Any idea?

dpgeorge, 2019-11-28:
> I'm getting `error freezing mpy` when trying to freeze some module with Viper code.

Ah, the qstr linking process was not supported for Xtensa architecture in mpy-tool.py.  Should be fixed by 7f24c297787cd4bd92647f407017acbecdb0f949

dpgeorge, 2021-06-12:
Native/viper has been working on esp32 for some time now, so closing this issue.

closed 2021-06-12 by dpgeorge