micropython/micropython #4609: QSPI Flash rare deadlock at boot

andrewleech, 2019-03-15:
Rarely (three times in last two days of heavy use) I've had my stm32f765 board lock up during startup (after reset, not cold power on).

This occurs while trying to read a file from working filesystem on the qspi chip fairly early in main.py

With my debugger, I found the deadlock to occur at this while loop:
https://github.com/micropython/micropython/blob/c9eb7eb449a246c607fa92a5313e3137d728fbcb/ports/stm32/qspi.c#L271

Resetting the board has always brought it back with it working next time, filesystem and file intact and read fine.

I haven't yet investigated the failure at all, just wanted to record it here while I have the location pinned down.

andrewleech, 2019-04-30:
@dpgeorge I cherry-picked out your spiflash/qspi updates from #4669 (d160f711,2771f17b,e5751b4e,8eb1e7b6,8b26ac69,9a9e6992) and tested them on our system, they appear to fix this particular problem perfectly. 

Aside from the occasional lock-up at boot reported here, we were getting occasional resets which were actually watchdog recovering from this issue, and in some configurations we would get that every 10m to 1h apart. They have not been seen all day with the patches in place however.

On a possibly-related note however, we are getting filesystem corruption (after file write) when running fat (or littlefs based on your other PR) on the qspi, even with usb mass storage disabled.

It usually is found by a filesystem read failure OSError shortly after a small file has been written. With fat filesystem, it often results in the fat table being corrupted, eg file name are garbled. Other times the file listing looks ok, but they fail when trying to read anything.

Have you seen anything like this, possibly related still to this MPU issue or the errata mentioned here: https://electronics.stackexchange.com/questions/356881/stm32f7-quad-spi-dummy-cycle-behavior

dpgeorge, 2019-05-02:
> On a possibly-related note however, we are getting filesystem corruption

The same issue could still be there, you just see it with lower probability.  Try the following:
1. In `qspi_memory_map()` just add `return;` to the start of the function so it does nothing.
2. At the end of `qspi_init()` add the following (to completely disable memory access to the QSPI region):
```c
    // Disable MPU
    __DMB();
    SCB->SHCSR &= ~SCB_SHCSR_MEMFAULTENA_Msk;
    MPU->CTRL = 0;

    // Config MPU region
    MPU->RNR = MPU_REGION_NUMBER1;
    MPU->RBAR = 0x90000000;
    MPU->RASR = MPU_INSTRUCTION_ACCESS_DISABLE  << MPU_RASR_XN_Pos   |
                MPU_REGION_NO_ACCESS            << MPU_RASR_AP_Pos   |
                MPU_TEX_LEVEL0                  << MPU_RASR_TEX_Pos  |
                MPU_ACCESS_NOT_SHAREABLE        << MPU_RASR_S_Pos    |
                MPU_ACCESS_NOT_CACHEABLE        << MPU_RASR_C_Pos    |
                MPU_ACCESS_NOT_BUFFERABLE       << MPU_RASR_B_Pos    |
                0x00                            << MPU_RASR_SRD_Pos  |
                MPU_REGION_SIZE_256MB           << MPU_RASR_SIZE_Pos |
                MPU_REGION_ENABLE               << MPU_RASR_ENABLE_Pos;
    __ISB();
    __DSB();
    __DMB();

    // Enable MPU
    MPU->CTRL = MPU_PRIVILEGED_DEFAULT | MPU_CTRL_ENABLE_Msk;
    SCB->SHCSR |= SCB_SHCSR_MEMFAULTENA_Msk;
    __DSB();
    __ISB();
```

andrewleech, 2019-05-02:
Thanks, I was just going to try something like that after discovering the "more answers" link at the bottom of https://community.st.com/s/question/0D50X00009XkgwcSAB/stm32f7-quadspi-and-the-data-cache-odd-behaviour

I had tried a return at the start of memory-map function, that alone didn't change it, but I haven't tried `MPU_REGION_NO_ACCESS` yet. I've been trying to speed up the issue, currently it usually fails after 1 - 2 hours... (thousands of file reads/writes) which makes for slow testing of changes. I'd much prefer to have a faster reproduction case before trialing more fixes.

I've been wondering it the issue can also be related to either speculative access or other interrupt related issues in between `QUADSPI->CCR` and `QUADSPI->AR` being set. 
Particularly related to a message on that forum by "chead1.5336873128029033E12":
```
I am seeing a similar problem but with writes to an actual SPI Flash. Occasionally the address 
register spontaneously changes to zero (you can actually see the value zero if you read 
back AR), and some of the time when this happens, address zero is sent on the bus instead 
of the proper address. I introduced a PRIMASK-based critical section starting before the 
write to AR and ending after BUSY went high, and this seems to eliminate the problem of 
the wrong address being sent. AR still sometimes changes to the wrong value, but apparently 
late enough that it doesn’t affect the signals on the bus.

I am also using data in system SRAM with the D-cache enabled.

Something that I tried that didn't help was using the MPU to force either the QUADSPI control 
registers or the memory mapping region (which I don’t use) to be strongly ordered, on the 
thought that either a random cache fill might be touching the memory mapping region and 
causing AR to be modified (which would be theoretically legal, albeit odd, since the memory 
mapping region is by default Normal-type memory), or that the cacheability attributes on the 
control registers might not be how ARMv7-M architecturally defines them 
(as Device-type memory).
```

I know you've already got a `raise_irq_pri()` call in the higher level to block USB, but perhaps much more needs to be blocked. the reporter above didn't say how much they blocked (perhaps full interrupts-off is worth it).

dpgeorge, 2019-05-02:
I haven't done testing, but I do suspect that completely disabling access to the 256MB QSPI region will fix the issue of AR being spontaneously changed.

I do have ways of setting up tests that trigger the speculative loads very often, so that testing only takes a few minutes.  But it's not trivial to do and has to be done carefully to be sure that one is actually tracking the correct bug/solution, so one can be sure the bug is gone in the full system.

dpgeorge, 2019-05-24:
Is this now resolved?

andrewleech, 2019-05-26:
The QSPI region changes does fix this initial issue as discussed, though the follow on far-more-rare corruption still occurs for me with the fixed MPU settings.

Switching to software qspi made it run for twice as long (46 hours) but it still failed in a similar way. Not sure if the longer runtime was just from the slower disk access.

Curiously, I had vfsstat() being printed out each cycle being run. The failure occurred immediately before the bfree count got to 4094. 
I've seen this in the past (some weeks back when I previously had the block details being printed). Perhaps my qspi flash settings are incorrect somewhere, or it's trying to overrun the end in some way, I'll look into them closer today hopefully. 

For reference, we've got a 256Mbit / 32MB chip, and the operation sequence running on repeat is basically:

`# end of previous cycle`

* writes a ~3KB cycle result file successfully (uos.sync() then reread with crc to confirm)
* print `{'blocks': 8183, 'bavail': 4094, 'bsize': 4096, 'frsize': 4096, 'bfree': 4095}` 

`# start of new cycle`

* reads a small json file
* takes a photo, does some processing
* fails to read a 2.4MB calibration file which it has read successfully in every cycle previously. Note: file has not be re-written, same file used to work: `OSError: Could not find the path`
* writes a ~3KB cycle result file successfully (uos.sync() then reread with crc to confirm)
* print`{'blocks': 8183, 'bavail': 4094, 'bsize': 4096, 'frsize': 4096, 'bfree': 4094}` 

`# cycles now stop due to failed test`
At this point, trying to reread the same large file continues to fail, including after reboot. 

I'll try filling up our flash today prior to starting a new test to see if that triggers fault sooner.

dpgeorge, 2019-05-27:
The fact that software QSPI has the same (or similar?) problem as hardware QSPI really does seem to rule out problems with speculation/MPU.  Which is good!

Are you using FAT-FS or LittleFS for the above run that fails?

>  print `{'blocks': 8183, 'bavail': 4094, 'bsize': 4096, 'frsize': 4096, 'bfree': 4095}`

This doesn't seem right, `bavail==bfree` always.  Is there a typo?

> fails to read a 2.4MB calibration file which it has read successfully in every cycle previously

From the sequence of events it sounds like this is due to the write of the 3KB result file in the previous run.

My suggestions:
- At each cycle capture the full contents of the block device being used and dump them somewhere (eg to a big SD card, just copying blocks, or to serial like `pyb.USB_VCP(1)`).  You could save all of them, or just the last N (=10 for example).  Then after it fails you can hopefully very quickly reproduce the error by reloading the block device just before it failed.
- Artificially configure the block device to be much smaller to make it fail earlier.
- Split the block device into partitions, one for the 2.4MB calibration file and other critical files, and one for the 3KB result files for which you can afford to format the partition if there is an error.

andrewleech, 2019-05-27:
Yes it's great to rule out further MPU issues.
Pre-filling the flash to just before the magic bfree count then start it running has worked to reproduce, I got exactly the same failure at the same point. This is back on hardware qspi. So now I should be on the fast track to knocking out the issue!

And yes, that was a copy/reformat/paste error on the `bavail==bfree` fail above, they were always in sync.

These tests are on one single large FatFS running on the qspi. I turned littlefs off again to focus on one issue at a time.

I have wanted to split the filesystem into two partitions, one for static calibration files, the other for transient results. I couldn't see a way to quickly do that with the current structure of the spiflash block device to easily specify multiple mount points for different regions of flash? Is there a way to do this built in already, or just duplicate the mount code in `main.c`?

I was briefly running two mount points with the second partition configured with littlefs, that's probably how I'm going to want it long term anyway.

dpgeorge, 2019-05-27:
> These tests are on one single large FatFS running on the qspi.

Ok.  It'd be good to rule out errors at the low level by verifying the block device itself, doing lots of sequential and random block read/writes.  If that passes then the error could be in FatFS itself.

> I have wanted to split the filesystem into two partitions, one for static calibration files, the other for transient results
> ...
> Is there a way to do this built in already,

You can use both `MICROPY_HW_BDEV_xxx` and `MICROPY_HW_BDEV2_xxx` config sets to get two block devices, and they can (I'm pretty sure) be both on the same physical SPI flash.

andrewleech, 2019-05-27:
I saw the BDEV2 gets registered as a separate partition in storage.c, but couldn't see anywhere it would be mounted. Minor detail though, I'll get to that later.

It looks like I've found the issue, though untested yet. `qspi.c` 
``` c
STATIC void qspi_write_cmd_addr_data(void *self_in, uint8_t cmd, uint32_t addr, size_t len, const uint8_t *src) {
    (void)self_in;

    QUADSPI->FCR = QUADSPI_FCR_CTCF; // clear TC flag

    if (len == 0) {
        QUADSPI->CCR =
            0 << QUADSPI_CCR_DDRM_Pos // DDR mode disabled
            | 0 << QUADSPI_CCR_SIOO_Pos // send instruction every transaction
            | 0 << QUADSPI_CCR_FMODE_Pos // indirect write mode
            | 0 << QUADSPI_CCR_DMODE_Pos // no data
            | 0 << QUADSPI_CCR_DCYC_Pos // 0 dummy cycles
            | 0 << QUADSPI_CCR_ABMODE_Pos // no alternate byte
            | 2 << QUADSPI_CCR_ADSIZE_Pos // 24-bit address size
            | 1 << QUADSPI_CCR_ADMODE_Pos // address on 1 line
            | 1 << QUADSPI_CCR_IMODE_Pos // instruction on 1 line
            | cmd << QUADSPI_CCR_INSTRUCTION_Pos // write opcode
            ;
```
emphasis on
``` c 
            | 2 << QUADSPI_CCR_ADSIZE_Pos // 24-bit address size
```

I'm running into the issue when I'm half way though my (32MB) flash chip. 
16MB needs 25 bits. 
Do you know if there's any disadvantage to having the chip in 32bit addressing mode all the time?

ps. this also explains why the issue hasn't been seen on 2MB flash chips, etc.

dpgeorge, 2019-05-27:
> 16MB needs 25 bits

Ok, that would make sense.  It would be a separate command (for both read and write) to use a 32-bit address.  You'll need to check the datasheet for the SPI flash you use to find out what it is.

> Do you know if there's any disadvantage to having the chip in 32bit addressing mode all the time?

The cost of an extra byte for each transaction, and since addresses are on one line that's 8 cycles.

andrewleech, 2019-05-27:
Thanks, that's very helpful! 
I wasn't sure why just changing the ADSIZE from 24bit to 32bit wasn't working, updating the commands did the trick.

I've got a test running now, with the ADSIZE and command(s) being selectively updated if the address passed in is >= 16MB and initially it looks great, a couple of different test setups have both breezed past that previously failing boundary with no visible issues so far.  

One basic test was just writing a file at startup to fill the flash, with `mp_print` in c code on each erase/write command with address, that's what helped to narrow down the issue. I can not write past the previously failing address printout just fine.

I'll get the code cleaned up and in a MR tomorrow! Thanks again.

Josverl, 2025-03-08:
@andrewleech 
 It seems that this could be closed, or is there still value in keeping this open?

andrewleech, 2025-03-08:
Thanks for bringing it to be attention @Josverl yes it was resolved

closed 2025-03-08 by andrewleech