micropython/micropython #4611: ESP32 Port Uses Only One Core with FreeRTOS

oclyke, 2019-03-16:
I haven't investigated this very far yet, but I noticed that the sdkconfig files set ```CONFIG_FREERTOS_UNICORE=y```. I tried modifying this in order to pin a (pure c) task on core 1 but experienced some sort of crash. Ultimately it would be great to see the dual core feature of the ESP32 utilized, if only to allow core 0 to be in charge of the wireless communication stack while MicroPython runs on core 1. 

Here's my output recorded using the ESP IDF monitor.

```
WARNING: esp-idf git submodule components/esp32/lib may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
Toolchain path: /Users/me/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/me/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/cu.usbserial-1410 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (411) cpu_start: Pro cpu up.
I (411) cpu_start: Application information:
I (411) cpu_start: Compile time:     22:45:38
I (413) cpu_start: Compile date:     Mar 15 2019
I (418) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996
I (425) cpu_start: Starting app cpu, entry point is 0x40082848
0x40082848: _realloc_r at /Users/me/esp/esp-idf/components/newlib/syscalls.c:47

I (0) cpu_start: App cpu up.
I (435) heap_init: Initializing. RAM available for dynamic allocation:
I (442) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (448) heap_init: At 3FFB9E90 len 00026170 (152 KiB): DRAM
I (454) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (461) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (467) heap_init: At 40092944 len 0000D6BC (53 KiB): IRAM
I (473) cpu_start: Pro cpu start user code
I (44) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
/Users/me/partial_directory_name_cut_off_by_crash(?)
```

No rush. But I think this would be a very nice feature to add.

dpgeorge, 2019-03-18:
>  But I think this would be a very nice feature to add.

I agree!

Recently some build issues were fixed in fcace26d87a99a23e74461e0b59f740de9f40365 which means that dual core can now work.  But to get it working requires disabling the uPy threading support.  So there's some interaction between dual core and the `_thread` module which leads to the issues.

oclyke, 2019-03-18:
Awesome! I'll get more updated and plan to check it out soon.

RaphaelMaschinsen, 2019-03-26:
That's great news! 

I am trying to get dual core MicroPython working on the ESP32 since a while and i wasn't successful so far.

I tried to understand the way how it is done in the [MicroPython_ESP32_psRAM_LoBo](url) Repo that is based on this port and there are actually quite a lot of changes to the _mpthreadport.c_ file.

I tried to run the MicroPython Task with the freeRTOS function `xTaskCreateStaticPinnedToCore(&mp_task, "mp_task", mp_task_stack_len, NULL, CONFIG_MICROPY_TASK_PRIORITY, mp_task_stack, &mp_task_tcb, 0)` and initialized the necessary attributes beforehand in the code in the same way how Loboris did it. No matter what i do, i always run  into a similar Issue when trying to compile with the `CONFIG_FREERTOS_UNICORE=n` option. This is the output that i get when i'm trying to start MicroPython:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5096
load:0x40078000,len:9332
load:0x40080400,len:6216
entry 0x400806e8
I (430) cpu_start: Pro cpu up.
I (430) cpu_start: Application information:
I (430) cpu_start: Compile time:     15:28:20
I (432) cpu_start: Compile date:     Mar 26 2019
I (438) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996
I (444) cpu_start: Starting app cpu, entry point is 0x40082950
I (437) cpu_start: App cpu up.
I (455) heap_init: Initializing. RAM available for dynamic allocation:
I (462) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (468) heap_init: At 3FFB9BB8 len 00026448 (153 KiB): DRAM
I (474) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (480) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (487) heap_init: At 40093468 len 0000CB98 (50 KiB): IRAM
I (493) cpu_start: Pro cpu start user code
I (64) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
/Users/1/A/coremodule/esp-idf/components/freertos/queue.c:1441 (xQueueG/Users/1/A/coremodule/esp-idf/components/freertos/queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called aO ECr0r4 [Er600 on  ErO N

Backtrace: 0x40090fe3:0x3ffbc1f0 0x400912e1:0x3ffbc210 0x4008d600:0x3ffbc230 0x400f3729:0x3ffbc270 0x400f38c8:0x3ffbc290 0x4008d9c8:0x3ffbc2b0 0x4008e544:0x3ffbc2d0 0x4008cdd5:0x3ffbc300

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5096
load:0x40078000,len:9332
load:0x40080400,len:6216
entry 0x400806e8

```
However, when i compile with the same changes to the _main.c_ file but with the `CONFIG_FREERTOS_UNICORE=y` option activated, MicroPython runs normally.

dpgeorge, 2019-03-27:
I fixed dual core support in 815b79a8d0ff698bd577ad86280a599191dd5949 and enabled it by default in 92149c8a7954169285b2909012dc601c6e7cb0aa

closed 2019-03-27 by dpgeorge

RaphaelMaschinsen, 2019-03-27:
Wow, thats amazing! And btw thanks a lot for this great project!

gmli, 2019-03-27:
Hi, any idea on how to use the second core now ?

RaphaelMaschinsen, 2019-03-27:
I just recompiled with the newest commit and MicroPython is indeed by default running on both cores now, as you can see in the startup message:

```
I (430) cpu_start: Pro cpu up.
I (430) cpu_start: Application information:
I (430) cpu_start: Compile time:     13:08:31
I (432) cpu_start: Compile date:     Mar 27 2019
I (437) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996
I (444) cpu_start: Starting app cpu, entry point is 0x40082950
I (0) cpu_start: App cpu up.
I (454) heap_init: Initializing. RAM available for dynamic allocation:
I (461) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (467) heap_init: At 3FFB9BB8 len 00026448 (153 KiB): DRAM
I (473) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (480) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (486) heap_init: At 40093468 len 0000CB98 (50 KiB): IRAM
I (492) cpu_start: Pro cpu start user code
I (63) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```
If you want to dedicate a whole core to something else than MicroPython you have to look into the the ESP-IDF/freeRTOS functions [(https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/freertos-smp.html)](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/freertos-smp.html) the mentioned loboris port uses `xTaskCreateStaticPinnedToCore(...)`to pin a task to a specific core.

krekos, 2019-03-28:
@RaphaelMaschinsen  Hello, i want to ask how to enable two-core mode.  today i download latest micropython and compile it. but whent i boot board, in log i see it's started in single-core mode. Thank you.