micropython/micropython #4612: ucollections.deque constructor should accept empty list

goatchurchprime, 2019-03-17:
I spent some hours on the trying to get the deque constructor to work, and didn't sort it out until I searched this issues list and found some example code that was correct.  

Even though the [documentation](docs.micropython.org/en/latest/library/ucollections.html#ucollections.deque) says "*iterable* must be the **empty tuple**, and the new deque is created empty", I forgot that these words meant it had to be "()" and kept doing:

```
>>> from ucollections import deque
>>> q = deque([], 3)

Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
ValueError: 
```

I don't see why this has to be an error (it's not in CPython), or even why it's beneficial for the implementation to require the initializer to be empty.  

I can see that the class is designed to be compatible with the CPython code without sacrificing performance, which is why it has been decided to restrict this to fixed-length deques and only two of the methods.  (This means that Micropython code will always run under CPython, but not necessarily the other way round -- as long as you don't use the *flags* parameter.)  

But I think the line has been drawn too close by not allowing empty lists as well as empty tuples in constructor.

pfalcon, 2019-03-20:
> ucollections.deque constructor should accept empty list

Sorry, but why? Only because you:

> forgot that these words meant it had to be "()"

Did you consider other solutions, e.g. learn Python well enough to not mix up tuples and lists?

> I don't see why this has to be an error (it's not in CPython)

"**u**collections" module isn't in CPython either, so any assumptions related to CPython may be ungrounded.

> I can see that the class is designed to be compatible with the CPython code 

As everything else in MicroPython, it's designed to implement **subset** of CPython functionality. 

> without sacrificing performance

Not exactly, the complete situation is explained in  https://github.com/micropython/micropython/wiki/ContributorGuidelines

>  (This means that Micropython code will always run under CPython, but not necessarily the other way round

"Other way round" is never an aim, MicroPython differs from CPython in hundreds, possibly thousands things like that. But that's moot point here, because again, CPython doesn't have "ucollections" module. If you want to be compatible with CPython, you would need to use a real "collections" module. Or use CPython straight away.

goatchurchprime, 2019-04-20:
As I said, I was blocked on this one for hours, so I reported it.  Other people might have the same experience.  One tends to read documentation with some level of intuition, not every word literally.  

The normal train of logic with this constructor is to recognize the first argument does nothing and is just a placeholder, and was supposed to be empty.  In almost all such cases [] is no difference to ().

I have found the unit tests here:
    https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/tests/basics/deque2.py

And the implementation here:
    https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/py/objdeque.c#L45

```
/* Initialization from existing sequence is not supported, so an empty
       tuple must be passed as such. */
    if (args[0] != mp_const_empty_tuple) {
        mp_raise_ValueError(NULL);
    }
```

Now, if you were really trying to cut this code to the bone, you'd skip this test, because args[0] doesn't do anything at all.  So raising an Error must be for the purpose of telling the programmer what is and isn't allowed.  The lack of an object "mp_const_empty_list" means that allowing empty lists is not so easy to test.  

I agree that there are thousands of differences with CPython, and the designers will have had to make tens of decisions per day on it.  No one expects all of these decisions to be obvious for to the designer and programmer-user at the same time.

dhouck, 2026-03-05:
While looking for existing mention of the discussion I started at https://github.com/orgs/micropython/discussions/18895, I came across this issue.  My current environment is stuck on 1.22.2 so I canʼt test, but according to [the docs](https://docs.micropython.org/en/v1.23.0/library/collections.html) this was fixed in 1.23.0; if thatʼs correct this should probably be closed.